YosysHQ/yosys #4470: Can't build from 0.42 tarball

spth, 2024-06-29:
### Version

The 0.42 tarball (can't build, so can't give yosys --version otuput)

### On which OS did this happen?

Linux

### Reproduction Steps

```
tar -xzf yosys-yosys-0.42.tar.gz
cd yosys-yosys-0.42/
LANG=C make
```

### Expected Behavior

yosys builds (like it does for me when using the 0.40 or 0.42 tarballs).

### Actual Behavior

yosys fails to build, giving this erroer message:
```
Initialize the submodule: Run 'git submodule update --init' to set up 'abc' as a submodule.
make: *** [Makefile:748: check-git-abc] Error 1
```

KrystalDelusion, 2024-07-01:
Related to #4403 and the same as https://github.com/YosysHQ/yosys/issues/4403#issuecomment-2156188699 wherein downloading via the release tar the abc directory is empty and the yosys directory is not a git repository, which also means the `git submodule update --init` command does not work.

KrystalDelusion, 2024-12-02:
As of v0.47 releases include a `[yosys.tar.gz](https://github.com/YosysHQ/yosys/releases/download/0.47/yosys.tar.gz)` which includes all sub repositories automatically.  Does this solve your issue?

spth, 2024-12-02:
Yes, 0.47 built fine for me.

KrystalDelusion, 2024-12-02:
Perfect, I will close this issue then :)

closed 2024-12-02 by KrystalDelusion