YosysHQ/yosys #4473: ABC: execution of command failed return code 134.

kareefardi, 2024-07-02:
### Version

Yosys 0.42+40 (git sha1 a739e21a5, g++ 11.4.0-1ubuntu1~22.04 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

[test-case.zip](https://github.com/user-attachments/files/16068298/test-case.zip)
Extract the attached zip. cd into the extracted folder and run the synthesis script `yosys -c synth.tcl`.

### Expected Behavior

Clean exit

### Actual Behavior

Looks like a crash

Ravenslofty, 2024-07-02:
First things first: running the testcase results in "yosys-abc: src/map/scl/sclLibUtil.c:77: void abc::Abc_SclHashCells(SC_Lib*): Assertion `*pPlace == -1' failed." because you're passing `-liberty $sky130_lib` twice in the arguments to `abc`; I'm pretty sure you didn't mean to do that.

The actually reported issue is instead "ERROR: ABC: execution of command ""/usr/local/bin/yosys-abc" -s -f /tmp/yosys-abc-TaAwGJ/abc.script 2>&1" failed: return code 139." after `stime -p`. 

After writing a wrapper script for abc that calls gdb, I get this:
```
ABC: Program received signal SIGSEGV, Segmentation fault.
ABC: 0x00005555559d42db in abc::Abc_SclTimeNtkPrint (p=p@entry=0x555557394bd0, fShowAll=fShowAll@entry=0,
ABC:     fPrintPath=fPrintPath@entry=1) at src/map/scl/sclSize.c:203
ABC: 203                    nLength = Abc_MaxInt( nLength, strlen(Abc_SclObjCell(pObj)->pName) );
ABC: #0  0x00005555559d42db in abc::Abc_SclTimeNtkPrint (p=p@entry=0x555557394bd0, fShowAll=fShowAll@entry=0,
ABC:     fPrintPath=fPrintPath@entry=1) at src/map/scl/sclSize.c:203
ABC: #1  0x00005555559d70cc in abc::Abc_SclTimePerformInt (pLib=pLib@entry=0x55555685b940, pNtk=pNtk@entry=0x555557407d70,
ABC:     nTreeCRatio=nTreeCRatio@entry=0, fUseWireLoads=fUseWireLoads@entry=0, fShowAll=fShowAll@entry=0,
ABC:     fPrintPath=fPrintPath@entry=1, fDumpStats=0) at src/map/scl/sclSize.c:690
ABC: #2  0x00005555559d73e0 in abc::Abc_SclTimePerform (pLib=0x55555685b940, pNtk=0x555557407d70, nTreeCRatio=nTreeCRatio@entry=0,
ABC:     fUseWireLoads=fUseWireLoads@entry=0, fShowAll=fShowAll@entry=0, fPrintPath=fPrintPath@entry=1, fDumpStats=0)
ABC:     at src/map/scl/sclSize.c:712
ABC: #3  0x00005555559a7472 in abc::Scl_CommandStime (pAbc=0x555556838200, argc=2, argv=0x55555687d780) at src/map/scl/scl.c:895
ABC: #4  0x00005555557a6c2b in abc::CmdCommandDispatch (pAbc=pAbc@entry=0x555556838200, pargc=pargc@entry=0x7ffffffed0e8,
ABC:     pargv=pargv@entry=0x7ffffffed0f0) at src/base/cmd/cmdUtils.c:157
ABC: #5  0x00005555557a078e in abc::Cmd_CommandExecute (pAbc=pAbc@entry=0x555556838200,
ABC:     sCommand=sCommand@entry=0x7ffffffed150 "stime -p;\n") at src/base/cmd/cmdApi.c:210
ABC: #6  0x000055555579e6d7 in abc::CmdCommandSource (pAbc=0x555556838200, argc=<optimized out>, argv=<optimized out>)
ABC:     at src/base/cmd/cmd.c:787
ABC: #7  0x00005555557a6c2b in abc::CmdCommandDispatch (pAbc=pAbc@entry=0x555556838200, pargc=pargc@entry=0x7fffffff5208,
ABC:     pargv=pargv@entry=0x7fffffff5210) at src/base/cmd/cmdUtils.c:157
ABC: #8  0x00005555557a078e in abc::Cmd_CommandExecute (pAbc=pAbc@entry=0x555556838200,
ABC:     sCommand=sCommand@entry=0x555556837e10 "source /tmp/yosys-abc-sK8f22/abc.script") at src/base/cmd/cmdApi.c:210
ABC: #9  0x00005555557e758a in abc::Abc_RealMain (argc=4, argv=<optimized out>) at src/base/main/mainReal.c:332
ABC: #10 0x00005555557e4e40 in main (argc=<optimized out>, argv=<optimized out>) at src/base/main/main.c:11
```

which may be useful when reporting this to upstream ABC.

Ravenslofty, 2024-07-02:
Reported upstream as berkeley-abc/abc#305.

kareefardi, 2024-07-03:
@Ravenslofty Thanks for your help. Yes the double `-liberty` wasn't intentional. 

> Also, though it's harmless, your script includes e.g. retime,-D,{D}, when {D} expands to include -D N already, resulting in abc getting -D -D N.
Thanks for noticing that as well.