YosysHQ/yosys #4477: ABC: execution of command failed return code 137.

drkome, 2024-07-05:
### Version

yosys 0.34

### On which OS did this happen?

Linux

### Reproduction Steps

Actually I use openlane. But during the synthesis phase, Yosystem gave an error, so I wanted to write it here. I created a normal flow to my processor. STEP 1 made me wait a long time. Then it gave this error.

`[INFO]: Running Synthesis (log: designs/KamerSoC/runs/RUN_2024.07.05_13.07.58/logs/synthesis/1-synthesis.log)...
[ERROR]: during executing yosys script /openlane/scripts/yosys/synth.tcl
[ERROR]: Log: designs/KamerSoC/runs/RUN_2024.07.05_13.07.58/logs/synthesis/1-synthesis.log
[ERROR]: Last 10 lines:

80. Executing ABC pass (technology mapping using ABC).

80.1. Extracting gate netlist of module `\SOC' to `/tmp/yosys-abc-lQoCQf/input.blif'..
Extracted 325976 gates and 415004 wires to a netlist network with 89026 inputs and 161154 outputs.

80.1.1. Executing ABC.
sh: line 1:   207 Killed                  "/build/bin/yosys-abc" -s -f /tmp/yosys-abc-lQoCQf/abc.script 2>&1
ERROR: ABC: execution of command ""/build/bin/yosys-abc" -s -f /tmp/yosys-abc-lQoCQf/abc.script 2>&1" failed: return code 137.
child process exited abnormally

[ERROR]: Creating issue reproducible...`

There is no general mention of the error in the log file. Or I don't understand exactly where the problem is. I'm very new to this. It was working normally when I ran another simple file. While running the system, I received this error on my own processor.

### Expected Behavior

The next step was waited for.

### Actual Behavior

It remained at Step 1. And it made an error.

Ravenslofty, 2024-07-05:
openlane should give you a reproducible or something, I think?

Ravenslofty, 2024-07-05:
That being said, the `Killed` line makes me wonder if somehow you ran out of free memory? That's the only situation I can think of where a process would get killed.

drkome, 2024-07-05:
> openlane should give you a reproducible or something, I think?

If you are talking about reproducible. If I need to share this file. This seems impossible for me right now. I am in a competition and I am not allowed to post the codes on the internet. I thought maybe someone knew because of the error code.

Ravenslofty, 2024-07-05:
no, the error code simply states that ABC was killed - again, presumably from running out of memory. it doesn't tell us why.

drkome, 2024-07-05:
> no, the error code simply states that ABC was killed - again, presumably from running out of memory. it doesn't tell us why.

I did some research too. Probably due to processor and ram. Because towards the end of the work, the system shows 90% RAM usage and 70% processor problems. I have 8gb ram. It seems like the system won't work. Just a strange thing, the system saw 94% RAM usage. But it gave an error when I used 85% RAM.

closed 2024-07-09 by drkome