YosysHQ/yosys #4478: Yosys Optimization Error: Optimization Stuck in PEEPOPT Pass

LoSyTe, 2024-07-06:
### Version

 Yosys 0.30+48

### On which OS did this happen?

Linux

### Reproduction Steps

Hello,

I encountered an issue while using Yosys to synthesize a Verilog file. 

I used the Verilog file rtl.v, and during the Yosys synthesis, the process got stuck for a long time.

I have minimized the test case as much as possible to help with troubleshooting. Attached is the Verilog file (rtl.v) that triggers this issue. I hope to get the community's help and attention.
[design_file.zip](https://github.com/user-attachments/files/16115186/design_file.zip)

### Expected Behavior

The synthesis process completes successfully within a reasonable time frame.

### Actual Behavior

The synthesis process gets stuck at the PEEPOPT pass and does not complete.

georgerennie, 2024-07-06:
This is another variant of the peepopt issues already seen. The newly merged #4455 should prevent this type of issue, but unfortunately because of the `int` precision issues `-offset*shiftadd_max_ratio` overflows and becomes negative, preventing the check from catching this.

#4448 (not yet merged) fixes this by limiting the widths so you can't get int overflow.

LoSyTe, 2024-07-07:
> This is another variant of the peepopt issues already seen. The newly merged #4455 should prevent this type of issue, but unfortunately because of the `int` precision issues [...]

Thank you very much for your response and help.

closed 2024-07-12 by LoSyTe