YosysHQ/yosys #4521: Synchronous-read memory recognized as asynchronous

gzz2000, 2024-08-05:
### Version

Yosys 0.43+11 (git sha1 49f547782, g++ 9.4.0-1ubuntu1~20.04.2 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Below is a design `main.v` with just a simple memory (synchronous read).
``` verilog
module main(clk, clk2, read_enable, write_enable, read_addr, write_addr, read_data, write_data);

   input clk, clk2, read_enable, write_enable;
   input [12:0] read_addr, write_addr;
   input [31:0] write_data;
   output reg [31:0] read_data;

   reg [31:0]   mem[2**13 - 1:0];
   always @(posedge clk) begin
      if(write_enable) mem[write_addr] <= write_data;
   end
   always @(posedge clk2) begin
      if(read_enable) read_data <= mem[read_addr];
   end

endmodule // main
```

Below is my memory map file `mem_map.txt` to be used with `memory_libmap`.
``` text
ram block $__RAM_SYNC_ {
    # Has 13 address bits
    abits 13;
    # The r/w width is 32 bits
    width 32;
    cost 1;
    # init any;

    port sw "W" {
        clock posedge;
        clken;
        wrtrans all old;
        optional;
    }
    port sr "R1" {
        clock posedge;
        clken;
        optional;
    }
    port sr "R2" {
        clock posedge;
        clken;
        optional;
    }
}

ram block $__RAM_ASYNC_ {
    # Has 13 address bits
    abits 13;
    # The r/w width is 32 bits
    width 32;
    cost 1;
    # init any;

    port sw "W" {
        clock posedge;
        clken;
        wrtrans all old;
[... 66 lines cut ...]
  - demux score: 0
  - cost: 34.000000
  - abits 13 dbits 32
  - chosen base width 32
  - swizzle 0 - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 1 - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 2 - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 3 - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 4 - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 5 - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 6 - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 7 - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 8 - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 9 - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 10 - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 11 - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 12 - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 13 - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 14 - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 15 - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 16 - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 17 - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 18 - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 19 - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 20 - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 21 - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 22 - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 23 - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 24 - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 25 - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 26 - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 27 - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 28 - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 29 - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 30 - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 31 - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
  - write port 0: port group W
  - read port 0: port group R1
mapping memory main.mem via $__RAM_ASYNC_
```

If I remove `$__RAM_ASYNC_` from memory map file and only keep `$__RAM_SYNC_`, it will fall back to FF emulation. But I expect it to use `$__RAM_SYNC_`.

```
# after removing $__RAM_ASYNC_ and re-run from beginning:
yosys> debug memory_libmap -lib mem_map.txt
 
4. Executing MEMORY_LIBMAP pass (mapping memories to cells).
Memory main.mem mapping candidates (post-geometry):
- logic fallback
  - cost: 262144.000000
Memory main.mem mapping candidates (after post-geometry prune):
- logic fallback
  - cost: 262144.000000
using FF mapping for memory main.mem
```

povik, 2024-08-05:
I am not sure if this is documented anywhere, but more passes need to run for the synchronous read ports to be recognized from RTL input. Please insert the following commands in between the `proc` and `memory_libmap` calls and see if that helps.

```
    opt_expr
    opt_dff
    opt_clean
    memory -nomap
```

The synthesis scripts within Yosys contain this sequence, directly or indirectly, which makes the recognition work.

povik, 2024-08-05:
Leaving some notes: The minimal sequence of commands required here is `opt_dff; opt_clean; memory_dff;`. With this script the read enable signal gets folded into the flip-flop, and in turn the flip-flop gets folded into the read port. We may want to document a "minimal recommended script" for memory inference, which may be `proc; opt_expr; opt_dff; opt_clean; memory -nomap`. Cc @KrystalDelusion

gzz2000, 2024-08-05:
I want to attach a further question, if the memory is implemented like this:
``` verilog
module main(clk, clk2, read_enable, write_enable, read_addr, write_addr, read_data, write_data);
   input clk, clk2, read_enable, write_enable;
   input [12:0] read_addr, write_addr;
   input [31:0] write_data;
   reg [12:0]   read_addr_reg;
   output [31:0] read_data;

   reg [31:0]   mem[2**13 - 1:0];
   always @(posedge clk) begin
      if(write_enable) mem[write_addr] <= write_data;
   end
   always @(posedge clk2) begin
      if(read_enable) read_addr_reg <= read_addr;
   end
   assign read_data = mem[read_addr_reg];

endmodule // main
```
Note the difference with my original post: now we store the read address synchronously and assign the memory read asynchronously. This is equivalent to a synchronous memory. If I synthesize the above in yosys, even `proc; opt_expr; opt_dff; opt_clean; memory -nomap` cannot prevent it from generating async memory blocks.

I expected this to work because this memory pattern is [documented here](https://yosyshq.readthedocs.io/projects/yosys/en/latest/using_yosys/synthesis/memory.html#synchronous-sdp-with-write-first-behavior-alternate-pattern).

povik, 2024-08-05:
> I expected this to work because this memory pattern is [documented here](https://yosyshq.readthedocs.io/projects/yosys/en/latest/using_yosys/synthesis/memory.html#synchronous-sdp-with-write-first-behavior-alternate-pattern).

There's a difference from the documented pattern in that you have two different clocks on the read port and the write port. The `memory_dff` pass, which is called from within `memory -nomap` and which is responsible for inferring the read port to be synchronous, explicitly checks for this case and stops the inference: https://github.com/YosysHQ/yosys/blob/01b99972b468c70110e900c2f4aa3614f5d51411/passes/memory/memory_dff.cc#L585-L591

As to why that check is there, I am not certain.

gzz2000, 2024-08-07:
I figured out why the check is there. I was wrong: the two code are actually not equivalent if the read/write clocks are different. Let's think about this scenario:
1. read clock rise, perform a read at address A, say the result is X.
2. write clock rise, write to address A value Y. At the same time, no read is performed.
3. look at the current ram output, without performing another read.
The normal synchronous memory will read out X, but the registered address asynchronous one will read out Y.

Nevertheless, transforming this memory to synchronous is still possible. The correct equivalent code should be:
``` verilog
module main(clk, clk2, read_enable, write_enable, read_addr, write_addr, read_data, write_data);
   input clk, clk2, read_enable, write_enable;
   input [12:0] read_addr, write_addr;
   input [31:0] write_data;
   reg [12:0]   read_addr_reg;
   output reg [31:0] read_data;

   reg [31:0]   mem[2**13 - 1:0];
   always @(posedge clk) begin
      if(write_enable) begin
        mem[write_addr] <= write_data;
        if (write_addr == read_addr_reg) read_data <= write_data;
      end
   end
   always @(posedge clk2) begin
      if(read_enable) begin
        read_addr_reg <= read_addr;
        read_data <= mem[read_addr];
      end
   end
endmodule // main
```
I wonder if we should make this transformation possible through yosys.

gzz2000, 2024-08-07:
I have a further concern. Running `check` after `memory_libmap` gives a lot of warnings like `Warning: Wire main.$auto$memory_libmap.cc:1950:emit_port$98 [0] is used but has no driver.`. I am worried that they might corrupt the design in further optimization.

These warnings persist even if I `read_verilog -lib` or `techmap -map` with a blackbox verilog model of the memory block. You can try with this verilog file:
``` verilog
(* blackbox *)
module \$__RAM_SYNC_ ( PORT_R1_CLK, PORT_R1_ADDR, PORT_R1_RD_DATA, PORT_R2_CLK, PORT_R2_ADDR, PORT_R2_RD_DATA, PORT_W_CLK, PORT_W_WR_EN, PORT_W_ADDR, PORT_W_WR_DATA );
   input PORT_R1_CLK;
   input [12:0] PORT_R1_ADDR;
   output [31:0] PORT_R1_RD_DATA;

   input PORT_R2_CLK;
   input [12:0] PORT_R2_ADDR;
   output [31:0] PORT_R2_RD_DATA;

   input PORT_W_CLK, PORT_W_WR_EN;
   input [12:0] PORT_W_ADDR;
   input [31:0] PORT_W_WR_DATA;
endmodule
```

and this map.txt file:
``` text

ram block $__RAM_SYNC_ {
    # Has 13 address bits
    abits 13;
    # The r/w width is 32 bits
    width 32;
    cost 1;
    # init any;

    port sw "W" {
        clock posedge;
        wrtrans all old;
    }
    port sr "R1" {
        clock posedge;
    }
    port sr "R2" {
        clock posedge;
    }
}
```

povik, 2024-08-07:
> I figured out why the check is there. ... Let's think about this scenario: ...

Right, that explains it. Thanks!

> Nevertheless, transforming this memory to synchronous is still possible. The correct equivalent code should be:
>
> ```verilog
> module main(clk, clk2, read_enable, write_enable, read_addr, write_addr, [...]

I am not convinced supporting this transformation is worth it. If the user wants the extra data path between the write port and the read port, they can write it into their input RTL and use a supported pattern for inferring the synchronous memory.

povik, 2024-08-07:
> I have a further concern. Running `check` after `memory_libmap` gives a lot of warnings like `Warning: Wire main.$auto$memory_libmap.cc:1950:emit_port$98 [0] is used but has no driver.`.

Yes, this looks like a bug in `memory_libmap`. Please open an extra ticket with the reproduction steps. If you agree I will close this one.

closed 2024-08-07 by povik

gzz2000, 2024-08-07:
Thanks povik. I opened a new ticket at https://github.com/YosysHQ/yosys/issues/4529