YosysHQ/yosys #4522: ERROR: Can't open ABC output file.

exhaust-create, 2024-08-05:
### Version

Yosys 0.27+3 (git sha1 b58664d44, x86_64-conda-linux-gnu-cc 11.2.0 -fvisibility-inlines-hidden -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -fdebug-prefix-map=/root/conda-eda/conda-eda/workdir/conda-env/conda-bld/yosys_1678231239250/work=/usr/local/src/conda/yosys-0.27_4_gb58664d44 -fdebug-prefix-map=/home/cmj/openroad/conda-yosys=/usr/local/src/conda-prefix -fPIC -Os -fno-merge-constants)

### On which OS did this happen?

Linux

### Reproduction Steps

I am using Hammer to utilize Yosys to synthesize TinyRocket with ASAP7. However, an error occurred and the detailed information is shown at https://github.com/ucb-bar/chipyard/issues/1936. What should I do?

Feel free to ask for more related files or information.

### Expected Behavior

Successfully synthesize TinyRocket.

### Actual Behavior

ERROR: Can't open ABC output file, because of the lack of output.blif.

Ravenslofty, 2024-08-06:
Can you reproduce this with a version of Yosys which isn't over a year old?

exhaust-create, 2024-08-07:
> Can you reproduce this with a version of Yosys which isn't over a year old?

I have tried to update Yosys to **version** 0.41, but the same error came out again.

exhaust-create, 2024-08-07:
> Can you reproduce this with a version of Yosys which isn't over a year old?

I have updated the detailed information about the issue, you can check it further.