YosysHQ/yosys #4535: Yosys test segfaults

ACharlyR, 2024-08-13:
### Version

Yosys 0.44 (git sha1 80ba43d26, g++ 14.2.1 -march=x86-64 -mtune=generic -O2 -fno-plt -fexceptions -fstack-clash-protection -fcf-protection -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -flto=auto -fPIC -O3

### On which OS did this happen?

Linux

### Reproduction Steps

Git clone `https://aur.archlinux.org/packages/yosys-git`

Build and test

`makepkg -s`

The test also segfaults with the latest commit on main:

`Yosys 0.44+9 (git sha1 4b9f45273, g++ 14.2.1 -march=x86-64 -mtune=generic -O2 -fno-plt -fexceptions -fstack-clash-protection -fcf-protection -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -flto=auto -fPIC -O3)` 

I only run in to this issue on arch:

```
uname -r
6.10.4-arch2-1
```

There are no issues when following the same steps on another PCs running manjaro 5.10 and manjaro 6.6.

### Expected Behavior

No segfaults

### Actual Behavior

`yosys/tests/techmap/run-test.sh` segfaults. I have attached some debug infromation:

<details><summary>GDB traceback of coredump</summary>
<p>

```
$ gdb /tmp/yosys/src/yosys/yosys core.yosys.1005.56a485c5546940dd905acc28937a15c1.190234.1723536931000000
GNU gdb (GDB) 15.1
Copyright (C) 2024 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /tmp/yosys/src/yosys/yosys...
[New LWP 190234]

This GDB supports auto-downloading debuginfo from the following URLs:
  <https://debuginfod.archlinux.org>
Enable debuginfod for this session? (y or [n]) y
Debuginfod has been enabled.
To make this setting permanent, add 'set debuginfod enabled on' to .gdbinit.
Downloading separate debug info for /usr/lib/libpython3.12.so.1.0
Downloading separate debug info for /usr/lib/libm.so.6                                                                                                        
Downloading separate debug info for /usr/lib/libboost_python312.so.1.83.0                                                                                     
Downloading separate debug info for /usr/lib/libboost_system.so.1.83.0                                                                                        
Downloading separate debug info for /usr/lib/libboost_filesystem.so.1.83.0                                                                                    
[... 68 lines cut ...]
<details><summary>ldd</summary>
<p>

```
ldd /tmp/yosys/src/yosys/yosys
    linux-vdso.so.1 (0x00007a274b6df000)
    libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x00007a274a000000)
    libpython3.12.so.1.0 => /usr/lib/libpython3.12.so.1.0 (0x00007a2749800000)
    libm.so.6 => /usr/lib/libm.so.6 (0x00007a2749f11000)
    libboost_python312.so.1.83.0 => /usr/lib/libboost_python312.so.1.83.0 (0x00007a274a318000)
    libboost_system.so.1.83.0 => /usr/lib/libboost_system.so.1.83.0 (0x00007a274a313000)
    libboost_filesystem.so.1.83.0 => /usr/lib/libboost_filesystem.so.1.83.0 (0x00007a274a2ef000)
    libreadline.so.8 => /usr/lib/libreadline.so.8 (0x00007a274a297000)
    libffi.so.8 => /usr/lib/libffi.so.8 (0x00007a274a28c000)
    libz.so.1 => /usr/lib/libz.so.1 (0x00007a2749ef8000)
    libtcl8.6.so => /usr/lib/libtcl8.6.so (0x00007a274964f000)
    libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x00007a2749eca000)
    libc.so.6 => /usr/lib/libc.so.6 (0x00007a274945e000)
    /lib64/ld-linux-x86-64.so.2 => /usr/lib64/ld-linux-x86-64.so.2 (0x00007a274b6e1000)
    libboost_atomic.so.1.83.0 => /usr/lib/libboost_atomic.so.1.83.0 (0x00007a2749ec0000)
    libncursesw.so.6 => /usr/lib/libncursesw.so.6 (0x00007a2749e51000)
``` 

</p>
</details>

povik, 2024-08-13:
> no issues when following the same steps on another PCs running manjaro 5.10 and manjaro 6.6.

I take it those are Arch versions.

The crash is strange, we are crashing within the map erase call inside the `Wire` destructor. This is related to the Python binding, so disabling that may avoid the crash, but still, it's not clear to me how we could be misusing the `std::map` container to cause the crash.

ACharlyR, 2024-08-13:
> > no issues when following the same steps on another PCs running manjaro 5.10 and manjaro 6.6.
> 
> I take it those are Arch versions.
> 
These are Linux kernel versions, I forgot to specify that.

widlarizer, 2024-08-29:
Successfully reproduced in archlinux docker on up to date nixos. Hard to get debuginfod to work in docker though so I'll try on my manjaro laptop

widlarizer, 2024-08-29:
Manjaro laptop is a bit out of date and didn't reproduce. `yosys -V`: `Yosys 0.44+60 (git sha1 0fc5812dc, g++ 14.1.1 -march=x86-64 -mtune=generic -O2 -fno-plt -fexceptions -fstack-clash-protection -fcf-protection -fPIC -O3)`

widlarizer, 2024-08-29:
To reproduce, it's sufficient to build with gcc, LTO, PYOSYS, and then run `yosys -p "design -save orig"`

widlarizer, 2024-08-30:
I can't put more time into this, but the core issue is that DesignPass and all_designs get destroyed in the "wrong" order. It seems resolving it isn't very easy, register.cc may have to be involved, as well as `misc/py_wrap_generator.py`. For AUR maintainers, I don't have a better solution than disable pyosys. Disabling tests would allow yosys to be used though it would still double free and crash at exit in cases where designs are saved