YosysHQ/yosys #4556: Support for Relative Paths in Yosys scripts

RCoeurjoly, 2024-08-21:
### Feature Description

#### Feature Request: Support for Relative Paths in Yosys scripts

##### Problem Description
When executing a Yosys script (`.ys` file) that contains relative file paths, Yosys fails if it is not executed from the same directory where the script is. This creates issues when trying to run scripts from different directories.

##### Example Scenario
Consider the following situation:
- Directory: `dir/`
  - Files:
    - `my_script.ys`
    - `design.v`

Contents of `my_script.ys`:
`read_verilog design.v`
Contents of design.v (although not relevant for this feature):
```verilog
module pass_through(
    input wire a,
    output wire y
);

assign y = a;

endmodule
```

If I execute Yosys from outside the `dir` directory, like this:

`yosys -p "script dir/my_script.ys"`

I get the following error:

> /----------------------------------------------------------------------------\
> |  yosys -- Yosys Open SYnthesis Suite                                       |
> |  Copyright (C) 2012 - 2024  Claire Xenia Wolf <claire@yosyshq.com>         |
> |  Distributed under an ISC-like license, type "license" to see terms        |
> \----------------------------------------------------------------------------/
> Yosys 0.44+20 (git sha1 e4c8bb0ac, g++ 11.4.0-1ubuntu1~22.04 -Og -fPIC)
> 
> -- Running command `script dir/my_script.ys' --
> 
> -- Executing script file `dir/my_script.ys' --
> ERROR: Can't open input file `design.v` for reading: No such file or directory
> 

This happens because Yosys attempts to read design.v from the current directory, not relative to the script's location.

#### Feature Request
Add a new option to the script command, such as -relative_path.

When -relative_path is used, Yosys should prepend the relative path of the .ys script to all relative file paths mentioned in the script.

Note that this should affect only relative file paths in the script, not absolute paths.
##### Example Command
`yosys -p "script -relative_path dir/my_script.ys"`

povik, 2024-08-21:
I would find having a way to make sure paths from the script are interpreted relative to the script location very useful, though I would prefer this was controlled by the content of the script rather than passing a flag externally. We already place special meaning on the `+/` prefix in paths, maybe we could invent another prefix for this.

RCoeurjoly, 2024-08-21:
> I would find having a way to make sure paths from the script are interpreted relative to the script location very useful, though I would prefer this was controlled by the content of the script [...]

Yeah, that would be a good way also to implement this.

closed 2025-08-20 by RCoeurjoly