YosysHQ/yosys #4560: Yosys is sensitive to the hash function

widlarizer, 2024-08-22:
### Version

All

### On which OS did this happen?

Linux

### Reproduction Steps

See #4559

### Expected Behavior

Tests pass, no matter the hash function

### Actual Behavior

`tests/techmap/dfflibmap.ys` fails when the hash function is changed

georgerennie, 2024-08-27:
Hunting this one down, ive minimized a reproducer to where yosys reads in the `dffsr` model in techmapping.

on the following script
```bash
read_verilog <<EOT
module dffsr(input CLK, D, CLEAR, PRESET, output reg Q);

always @(posedge CLK, posedge CLEAR, posedge PRESET)
	if (CLEAR)
		Q <= 0;
	else if (PRESET)
		Q <= 1;
	else
		Q <= D;

endmodule
EOT

proc -noopt

opt_expr -mux_bool
clean -purge
write_verilog -noattr
```

mainline yosys produces the following:
```verilog
assign _0_ = CLEAR ? 1'h0 : PRESET;
always @(posedge CLK, posedge _0_, posedge CLEAR)
  if (CLEAR) Q <= 1'b0;
  else if (_0_) Q <= 1'b1;
  else Q <= D;
```

whereas yosys with the new hash produces:
```verilog
assign _0_ = PRESET ? 1'h0 : CLEAR;
always @(posedge CLK, posedge PRESET, posedge _0_)
  if (_0_) Q <= 1'b0;
  else if (PRESET) Q <= 1'b1;
  else Q <= D;
```

In the first, clear takes priority over preset (as specified in the original verilog), whereas somehow in the version with the new hash the priority has been inverted and preset takes priority over clear. This change is introduced by `proc -noopt` (i added the further optimizations after just to make the verilog clearer but confirmed the behaviour appears before them).

georgerennie, 2024-08-27:
`proc_arst; proc_dff` seems sufficient.

From what I can tell, running `proc_arst` generates the following:
```
process $proc$<<EOT:3$1
  assign $0\Q[0:0] \Q
  assign $0\Q[0:0] \D
  sync high \CLEAR
    update \Q 1'0
  sync high \PRESET
    update \Q 1'1
  sync posedge \CLK
    update \Q $0\Q[0:0]
end
```

The priority seems to be encoded explicitly in the ordering of the sync rules (I don't know much about sync rules but this seems reasonable). The issue is then in `proc_dff`. These sync rules are [put into a `std::map`](https://github.com/YosysHQ/yosys/blob/main/passes/proc/proc_dff.cc#L222) from trigger signal to sync rule. This loses the ordering of the rules, and thus when they are subsequently processed in `gen_dffsr_complex`, they can be considered in the wrong order.

widlarizer, 2024-08-27:
`std::map` uses default `std::less` which uses `RTLIL::SigSpec::operator<` which uses the `hash_` built with the hash function. This shows that currently `proc_dff` in some cases has behavior depending on a well-defined but arbitrary ordering. The linked PR prevents this by processing the SigSpecs in insertion order, which corresponds to the design order if I'm right

PR #4569 (2024-09-02): proc_dff: respect sync rule priorities when generating complex dffsrs
This fixes #4560, where previously the order that sync rules were processed in depended on the order they were pulled out of a std::map. This PR changes this to process them in the order they are found in, respecting the priorities among the async signals.

This is an alternative to #4568 that should fix the issues highlighted there (I've tested the example I left in the replies there and it seems to give the correct result now).

It still makes the optimizations made before, such as merging together control signals if all s [... cut]
commit bdb5d455 (2024-09-02): proc_dff: respect sync rule priorities when generating complex dffsrs

* This fixes #4560, where previously the order that sync rules were
  processed in depended on the order they were pulled out of a std::map.
  This PR changes this to process them in the order they are found in,
  respecting the priorities among the async signals
closed 2024-09-02 by widlarizer