YosysHQ/yosys #4574: Seemingly shareable memory ports don't get merged

danderson, 2024-08-30:
### Version

Yosys 0.44 (git sha1 80ba43d26264738c93900129dc0aab7fab36c53f, g++ 13.3.0 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

Implement a basic memory, but represented as an array of 1-bit regs instead of the word width:

```systemverilog
module bitram2x1(input CLK,
                 input [1:0] AD,
                 input [1:0] DI,
                 input WE,
                 output reg [1:0] DO);
  (* no_rw_check *)
  reg ram [7:0];

  integer word_addr = AD*2;

  always @(posedge CLK) begin
    if (WE) begin
      for (integer i = 0; i < 2; i=i+1) begin
         ram[word_addr+i] <= DI[i];
      end
    end

    for (integer i = 0; i < 2; i=i+1) begin
      DO[i] <= ram[word_addr+i];
    end
  end
endmodule
```

Load into yosys and do some basic synth passes. Both these scripts produce the issue. First attempt with just basic generic synthesis:

```
read_verilog -sv weirdram.v; prep; dump -o weirdram.rtlil
```

Then with explicitly trying all the resource sharing passes:

```
read_verilog -sv weirdram.v; proc; flatten; wreduce;;; share; memory_share; opt_merge; memory_collect; dump -o weirdram.rtlil
```

I also tried removing `no_rw_check` and varying the module's port widths. This behavior seems consistent in all the variations I tried, up to 18b wide DI/DO.

### Expected Behavior

I expected both the 2 one-bit write ports and the 2 one-bit read ports to get merged, resulting in a `$mem_v2` with a single read port and a single write port, both 2 bits wide.

### Actual Behavior

The ports remain un-merged, the `$mem_v2` shows 2 one-bit read ports and two one-bit write ports. `show` confirms that the pre- and post-logic to generate multiple copies of addresses and mux single bits back together remains in the design.

Notable in the log is that `memory_collect` reports that the ports cannot be shared by address, despite always accessing an adjacent pair of bits. SAT sharing also says that sharing is impossible, but I failed to find a way to make it describe the failure further, so that may be expected.

```
7. Executing MEMORY_SHARE pass (consolidating $memrd/$memwr cells).
Consolidating read ports of memory bitram2x1.ram by address:
Consolidating write ports of memory bitram2x1.ram by address:
Consolidating write ports of memory bitram2x1.ram using sat-based resource sharing:
  Checking group clocked with posedge \CLK, width 1: ports 0, 1.
  Common input cone for all EN signals: 1 cells.
  Size of unconstrained SAT problem: 9 variables, 18 clauses
  According to SAT solver sharing of port 0 with port 1 is not possible.
```

KrystalDelusion, 2024-09-02:
Using arithmetic when indexing limits optimisations so try to avoid it if you can.
I can get this to pass with the following Yosys script on latest main (0fc5812dc):
```
read_verilog -sv << EOT
module bitram2x1(input CLK,
                 input [1:0] AD,
                 input [1:0] DI,
                 input WE,
                 output reg [1:0] DO);
  (* no_rw_check *)
  reg ram [7:0];

  always @(posedge CLK) begin
    if (WE) begin
      for (integer i = 0; i < 2; i=i+1) begin
         ram[{AD,i}] <= DI[i];
      end
    end

    for (integer i = 0; i < 2; i=i+1) begin
      DO[i] <= ram[{AD,i}];
    end
  end
endmodule
EOT

hierarchy -top bitram2x1; proc; flatten; memory -nomap;
select -assert-count 1 t:$mem_v2 r:RD_PORTS=1 %i
```
Note the `{AD,i}` instead of using an addition in the indexing.

Adding `opt_mem_widen` to your script (`proc; flatten; wreduce;;; share; memory_share; opt_mem_widen; opt_merge; memory_collect`) instead of the full `memory -nomap` also works.

danderson, 2024-09-02:
Thank you! I didn't know arithmetic here had such a dramatic impact, I'll keep that in mind in future. I'll also play with `opt_mem_widen` to get a feel for when it helps.

It sounds like this is behaving as expected then, given currently available analysis passes. I'll close this bug in a couple days, unless y'all want to keep it around for some reason.

closed 2024-09-07 by danderson