YosysHQ/yosys #4575: New `$check` flavors: `warning` and `error`

povik, 2024-08-30:
### Feature Description

Add new `$check` flavors `warning` and `error` to which a frontend can lower any calls to `$error` and `$warning` in procedural SystemVerilog code.

The difference of `error` from `assert` is that there's no `EN`/`A` distinction. When the error cell is enabled (`EN` is high) the check has failed, while for an assert the combination of `EN && !A` signifies failure.

Having the new flavors allows to preserve more of the original RTL intent in the netlist. As an alternative a frontend can lower `$error` to assert checks with `A` at constant zero.

whitequark, 2024-08-30:
> As an alternative a frontend can lower `$error` to assert checks with `A` at constant zero.

I would prefer either this option, or lowering to a flavor `error` with `A` at constant zero (for roundtripping).

I find it undesirable to make the semantics of the cell depend on the flavor. That's why it's called "flavor" and not "type" (and isn't something reflected by the type of cell): it should be transparent to consumers.

/cc @jix

povik, 2024-08-30:
> lowering to a flavor `error` with `A` at constant zero (for roundtripping).

Yes, flavor `error` should have `A` at constant zero. At least in practice though I would prefer if it was part of the definition of the flavor.

> I find it undesirable to make the semantics of the cell depend on the flavor. That's why it's called "flavor" and not "type"

Is that the case with the existing flavors, though? I am not sure I would say `assert` and `cover` have the same semantics. The first fires when `EN && !A`, the second when `EN && A`

whitequark, 2024-08-30:
> Is that the case with the existing flavors, though? I am not sure I would say `assert` and `cover` have the same semantics. The first fires when `EN && !A`, the second when `EN && A`

Oh, you're right. I think this was my original design intent but it ended up changing in all the discussion.

In this case I no longer have a preference.