YosysHQ/yosys #4586: Github tag style

hpretl, 2024-09-07:
### Feature Description

Recently, the tag style for Yosys changed from `yosys-0.44` to `0.45`. Is this the new style going forward, or is this a typo? In either case, please strive for a consistent tag style between `yosys`, `sby`, `mcy`, and `eqy`.

whitequark, 2024-09-07:
#4578

whitequark, 2024-09-07:
For the record, I would have personally preferred `vX.Y` rather than `X.Y` because the former allows listing all tags with `git tag v*` and the latter requires you to do the far more error prone `git tag '*.*'`. (There is no way to write a glob pattern matching a variable digit amount number.)

KrystalDelusion, 2024-09-08:
ReadtheDocs supposedly follows PEP440, which does allow for the [preceding v character](https://peps.python.org/pep-0440/#preceding-v-character), although I'm not sure how exactly that fits with the [existing docs versioning](https://readthedocs.org/projects/yosys/).  I was actually under the impression that the plan was to tag releases with both the `yosys-0.xx` style _and_ the `0.xx` for docs.

whitequark, 2024-09-08:
> although I'm not sure how exactly that fits with the [existing docs versioning](https://readthedocs.org/projects/yosys/).

It strips the `v` prefix.

KrystalDelusion, 2024-09-08:
Perfect.  I'm pro v prefix then

KrystalDelusion, 2024-12-11:
This should now be the case as of [`v0.48`](https://github.com/YosysHQ/yosys/releases/tag/v0.48).  Note also the v prefix.

closed 2024-12-11 by KrystalDelusion

hpretl, 2024-12-12:
Yep, I realized already yesterday with a failing build. :-) At least now Yosys and EQY/etc. are harmonized 👍