YosysHQ/yosys #4590: Synthesis with synth_xilinx crashes in TECHMAP pass

marzoul, 2024-09-09:
### Version

Yosys 0.45+106

### On which OS did this happen?

Linux

### Reproduction Steps

See attached archive. It contains the input verilog file, the Makefile, and the log of last Yosys run.
Just type `make` to reproduce.

The design has been reduced to as much as possible to reproduce the crash reliably.
The original design was VHDL, converted to Verilog with the tool ghdl, this is why the verilog is not easily readable.
Overall the datapath that remains is a series of FIFOs and a small read-only memory in parallel.
The issue is not reproduced with less memories hence the design is not that small. 

Complete version of Yosys as reported by `yosys --version` :
`Yosys 0.45+106 (git sha1 982fade0d, g++ 14.2.1 -march=x86-64 -mtune=generic -O2 -fno-plt -fexceptions -fstack-clash-protection -fcf-protection -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -ffile-prefix-map=/home/prostboa/Dev/AUR/yosys-git/src=/usr/src/debug/yosys-git -flto=auto -fPIC -O3)`

Yosys is freshly compiled from sources, from Archlinux AUR package yosys-git.

### Expected Behavior

Yosys is expected to finish the synthesis and emit the resulting json file without error.

### Actual Behavior

Yosys emits an error in TECHMAP pass and exits with error code zero (it should also be non-zero in such error case).

marzoul, 2024-09-09:
For whatever reason I could not attach the small archive (77kB).
Here is a public link : https://cloud.univ-grenoble-alpes.fr/s/YFwoXCHti7sTasM

georgerennie, 2024-09-10:
Minimized reproducer:
```
read_ilang <<EOT
autoidx 1
module \nnawaq_axi3
  wire output 1 \out
  cell $_MUX16_ \mux
    connect \A 1'x
    connect \B 1'x
    connect \C 1'x
    connect \D 1'x
    connect \E 1'x
    connect \F 1'x
    connect \G 1'x
    connect \H 1'x
    connect \I 1'x
    connect \J 1'x
    connect \K 1'x
    connect \L 1'x
    connect \M 1'x
    connect \N 1'x
    connect \O 1'x
    connect \P 1'x
    connect \S 1'x
    connect \T 1'x
    connect \U 1'x
    connect \V 1'x
    connect \Y \out
  end
end
EOT

techmap -map +/xilinx/cells_map.v -D MIN_MUX_INPUTS=5
```

KrystalDelusion, 2025-03-27:
```
// Trim off any leading 1'bx -es in A
else if (_TECHMAP_CONSTMSK_A_[A_WIDTH-1] && _TECHMAP_CONSTVAL_A_[A_WIDTH-1] === 1'bx) begin
  localparam A_WIDTH_new = A_WIDTH_trimmed(A_WIDTH-1);
  \$__XILINX_SHIFTX  #(.A_SIGNED(A_SIGNED), .B_SIGNED(B_SIGNED), .A_WIDTH(A_WIDTH_new), .B_WIDTH(B_WIDTH), .Y_WIDTH(Y_WIDTH)) _TECHMAP_REPLACE_ (.A(A[A_WIDTH_new-1:0]), .B(B), .Y(Y));
end
```
seems to be the relevant piece of code, the problem being that it doesn't handle the case where the input is *only* `'x`.

KrystalDelusion, 2025-03-27:
Absent the context (the link to the archive no longer works), I'm not sure if the all `'x` mux has come from the source or been generated along the way; so while the `xilinx/cells_map.v` is the point of failure I'm not sure if that's the best place to fix it.

marzoul, 2025-03-28:
Thank you for having spent time on this.
I will look at why the archive is no longer available and try to reproduce with recent design and yosys tool.

marzoul, 2025-03-30:
I can reproduce with freshly generated RTL and up to date ghdl and Yosys tools. The design simulates OK (vhdl source), and without the `-widemux 5` option the synthesized design does work on board. All looks like the source is fine.

I will reduce the design to something manageable.

In the meantime, I can suggest a potential source for the issue : the RTL implementation of involved components is highly configurable, so it is possible that some unused parts of logic remain. If there is only one root cause involved in this issue, it could be that Yosys ends up trying to techmap a (unused) mux with unconnected/uninitialized inputs.
If this is the case, this relatively common implementation pattern should not make Yosys crash ; instead it would be preferrable that Yosys propagate the `x` and remove the involved logic.
A warning message indicating where this issue comes from would also be super useful :)

marzoul, 2025-03-31:
Reproducer in verilog :
```verilog
module top (
	input  clk,
	output [7:0] rddata
);

  reg [31:0] index;

  always @(posedge clk)
    index <= index + 1;
  initial
    index = 32'b0;

  reg [7:0] rom[33:0];
  initial begin
    rom[33] = 8'b00000000;
    rom[32] = 8'b00000001;
    rom[31] = 8'b00000010;
    rom[30] = 8'b00000011;
    rom[29] = 8'b00000100;
    rom[28] = 8'b00000101;
    rom[27] = 8'b00000110;
    rom[26] = 8'b00000111;
    rom[25] = 8'b00001000;
    rom[24] = 8'b00001001;
    rom[23] = 8'b00001010;
    rom[22] = 8'b00001011;
    rom[21] = 8'b00001100;
    rom[20] = 8'b00001101;
    rom[19] = 8'b00001110;
    rom[18] = 8'b00001111;
    rom[17] = 8'b00010000;
    rom[16] = 8'b00010001;
    rom[15] = 8'b00010010;
    rom[14] = 8'b00010011;
    rom[13] = 8'b00010100;
    rom[12] = 8'b00010101;
    rom[11] = 8'b00010110;
    rom[10] = 8'b00010111;
    rom[ 9] = 8'b00011000;
    rom[ 8] = 8'b00011001;
    rom[ 7] = 8'b00011010;
    rom[ 6] = 8'b00011011;
    rom[ 5] = 8'b00011100;
    rom[ 4] = 8'b00011101;
    rom[ 3] = 8'b00011110;
    rom[ 2] = 8'b00011111;
    rom[ 1] = 8'b00100000;
    rom[ 0] = 8'b00100001;
    end

  assign rddata = rom[index];

endmodule
```
Launch with :
```sh
yosys -l yosys.log -p "synth_xilinx -arch xc7 -top top -widemux 5 ; write_json top.json" top.v
```
Error message :
```
2.42.39. Executing OPT_EXPR pass (perform const folding).
Optimizing module $paramod$constmap:843fb74262f5d2d0f71bbae0f21a2bc38ba47b58$paramod$ff678c476fc5f92d51299d5befc9ff8e1fd3ef3a\_90_shift_shiftx.
<suppressed ~4 debug messages>
Removed 2 unused cells and 10 unused wires.
Using template $paramod$constmap:843fb74262f5d2d0f71bbae0f21a2bc38ba47b58$paramod$ff678c476fc5f92d51299d5befc9ff8e1fd3ef3a\_90_shift_shiftx for cells of type $shiftx.
<suppressed ~24 debug messages>
/usr/bin/../share/yosys/xilinx/cells_map.v:151: ERROR: Signal `\A' with invalid width range -1!
```

marzoul, 2025-03-31:
So it's not about uninitialized or unused things : some optim passes do produce invalid netlist or expressions with `-widemux 5` and  a ROM of size >32.

KrystalDelusion, 2025-04-01:
So I've replaced the `fine` step with the commands it runs internally and found where the failure is introduced:
```
read_verilog top.v
synth_xilinx -arch xc7 -top top -widemux 5 -run :fine
simplemap t:$mux
muxcover -nodecode -mux2=100 -mux8=399 -mux16=400
opt -full #<--- introduces the failure
xilinx_srl -variable -minlen 3
techmap -map +/techmap.v -D LUT_SIZE=6 -D MIN_MUX_INPUTS=5 -map +/xilinx/mux_map.v -map +/xilinx/arith_map.v
opt -fast
synth_xilinx -arch xc7 -top top -widemux 5 -run map_cells:
```
Removing the `opt -full` there (or replacing it with a regular `opt` lets the script complete as expected.  But when it is there it introduces undef inputs to the muxes and causes the failure.

I suspect the problem here is because `opt_expr -mux_undef`: `remove 'undef' inputs from $mux, $pmux and $_MUX_ cells` (which were themselves probably added by `opt_expr -undriven`: `replace undriven nets with undef (x) constants`), but that doesn't handle the wide cells.

Calling `opt -full` between the `simplemap` and `muxcover` also fixes the problem (by optimizing the `$mux` cells prior to `muxcover`), which results in a smaller circuit than the other options I suggested.  But in the `map_ffram` step (right before the `fine` step) there is this

```
if (widemux > 0) {
	run("opt -fast -mux_bool -undriven -fine"); // Necessary to omit -mux_undef otherwise muxcover
						    // performs less efficiently
} else {
	run("opt -fast -full");
}
```
which for whatever reason seems to suggest that the `opt -full` before `muxcover` is undesirable for some reason...

marzoul, 2025-04-06:
I tested both PRs #4989 and #4994, both make the reproducer design above pass. Approches are different and possibly complementary in the way they strengthen the different parts of the code.