YosysHQ/yosys #4601: Importing init of memory is extremely slow

marzoul, 2024-09-17:
### Version

Yosys 0.45+106

### On which OS did this happen?

Linux

### Reproduction Steps

Complete Yosys version :
`Yosys 0.45+106 (git sha1 982fade0d, g++ 14.2.1 -march=x86-64 -mtune=generic -O2 -fno-plt -fexceptions -fstack-clash-protection -fcf-protection -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -ffile-prefix-map=/home/prostboa/Dev/AUR/yosys-git/src=/usr/src/debug/yosys-git -flto=auto -fPIC -O3)
`

Reproducer can't be attached due to size. Here is a public archive :
https://cloud.univ-grenoble-alpes.fr/s/f9ADZGf7sRpr9Ff

Init with zeros is done here for the sake of a simple reproducer. Actually for original design, init with arbitrary data is used.
What causes the issue if the init block with following template (generated by ghdl) :
```verilog
initial begin
    mem[65535] = 1'b0;
    mem[65534] = 1'b0;
    [...]
    mem[1] = 1'b0;
    mem[0] = 1'b0;
end
```

### Expected Behavior

Yosys should perform synthesis in very short time (for example less than 2 seconds)

### Actual Behavior

Yosys is taking at least several minutes to import the design (killed after 5 minutes)

marzoul, 2024-09-17:
Update : For unknown reasons, after 4 minuts Yosys gets killed by the operating system.
So it's not just a rapidity problem, there is a crash-type of issue involved.

epsilon537, 2025-01-14:
I'm seeing a similar issue. read_verilog of module below takes more than 8 minutes to complete. If I remove the initial block, it completes instantly.

```
module spram_byte_write_ram_wr_first #(
    parameter NB_COL = 4,  // Specify number of columns (number of bytes)
    parameter COL_WIDTH = 8,  // Specify column width (byte width, typically 8 or 9)
    parameter RAM_DEPTH = 32768  // Specify RAM depth (number of entries)
) (
    input  wire [clogb2(RAM_DEPTH-1)-1:0] addra,  // Address bus, width determined from RAM_DEPTH
    input  wire [ (NB_COL*COL_WIDTH)-1:0] dina,   // RAM input data
    input  wire                           clka,   // Clock
    input  wire [             NB_COL-1:0] wea,    // Byte-write enable
    output wire [ (NB_COL*COL_WIDTH)-1:0] douta   // RAM output data
);

  reg [(NB_COL*COL_WIDTH)-1:0] BRAM[RAM_DEPTH-1:0];
  reg [(NB_COL*COL_WIDTH)-1:0] ram_data = {(NB_COL * COL_WIDTH) {1'b0}};

  // The following code either initializes the memory values to a specified file or to all zeros to match hardware
  integer ram_index;
  initial for (ram_index = 0; ram_index < RAM_DEPTH; ram_index = ram_index + 1) BRAM[ram_index] = 0;

  generate
    genvar i;
    for (i = 0; i < NB_COL; i = i + 1) begin : byte_write
      always @(posedge clka)
        if (wea[i]) begin
          BRAM[addra][(i+1)*COL_WIDTH-1:i*COL_WIDTH] <= dina[(i+1)*COL_WIDTH-1:i*COL_WIDTH];
          ram_data[(i+1)*COL_WIDTH-1:i*COL_WIDTH] <= dina[(i+1)*COL_WIDTH-1:i*COL_WIDTH];
        end else begin
          ram_data[(i+1)*COL_WIDTH-1:i*COL_WIDTH] <= BRAM[addra][(i+1)*COL_WIDTH-1:i*COL_WIDTH];
        end
    end
  endgenerate

  assign douta = ram_data;
  ...
```
Here is the original module: 

[https://github.com/epsilon537/vera_wishbone/blob/boxlambda/fpga/source/spram_byte_write_ram_wr_first.v](https://github.com/epsilon537/vera_wishbone/blob/boxlambda/fpga/source/spram_byte_write_ram_wr_first.v
)

I'm using Yosys 0.44 on Linux.

marzoul, 2025-01-23:
I think the issue is there in the code : frontends/ast/simplify.cc:1799

```
		if (stage == 2 && children[i]->type == AST_INITIAL && current_ast_mod != this) {
			current_ast_mod->children.push_back(children[i]);
			children.erase(children.begin() + (i--));
			did_something = true;
		}
```
The many children of the the initial nide (a long list/vector) are extracted one by one and appended to a new vector of modified children. Children are removed from the beginning, so the overall complexity of this code is O(n^2).
Given that strictly all children will be moved (at least in our cases), there is no point removing children one by one.

I'm not knowledgeable at all about the code so I don't feel comfortable attempting to propose a fix...

marzoul, 2025-03-11:
I reduced the design to something much more manageable.
```verilog top.v
module top (
	input  clk,
	input  wen,
	input  [31:0] waddr,
	input  [31:0] wdata,
	input  ren,
	input  [31:0] raddr,
	output [31:0] rdata
);

	localparam SIZE = 16384;

	reg mem[SIZE];
	integer i;

	initial begin
		for(i=0; i<SIZE; i=i+1)
			mem[i] = 1'b0;
	end

	always @(posedge clk) begin
		if (ren)
			rdata = mem[raddr];
		if (wen)
			mem[waddr] <= wdata;
	end

endmodule
```

On my laptop, here are the stats for run time and RAM usage :
```
Size  4096 :   4 seconds, 150MB RAM
Size  8192 :   8 seconds, 200MB RAM
Size 16384 :  21 seconds, 310MB RAM
Size 32768 :  81 seconds, 530MB RAM
Size 65536 : 370 seconds, 970MB RAM
```

Digging with gdb, I found that most of the time seems spent in `SigSpec::remove2()` defined at `kernel/rtlil.cc:4166`
That one is called from `frontend/ast/genrtlil.cc` in function `removeSignalFromCaseTree()` near line 550.
The main loop iterates over **several hundred thousand actions**, each doing `remove2()` on vectors of bits and of chunks within `SigSpec` structures and also does linear scan of pattern chunks in there. This is heavy.

But maybe this is just a symptom.
The root cause seems to be about how the initial statements are applied to the memory.
Counting the log messages, these 2 ones (and similar in what they designate) appear in same number as the size of the memory :
```
No latch inferred for signal `\top.$memwr$\mem$top.v:26$1098_EN' from process `\top.$proc$top.v:0$17491'.
```
and
```
Removed top 19 address bits (of 32) from memory init port top.$meminit$\mem$top.v:26$10000 (mem).
```
So, it looks like the initial block is creating as many ports to the memory as there are lines in the memory. And later it tries to simplify these as legitimate logic, then these get removed along the flow. There is certainly a more efficient solution to handle simple initial blocks and avoid explosion of the number of ports.

whitequark, 2025-03-11:
This is a well known flaw in the Yosys Verilog frontend and it has been difficult to address because nobody really understands how the frontend works or is able to refactor it.

marzoul, 2025-03-11:
gasp. Looking at the code that was a bit what I feared. Thank you for the message. I'll see what I can do.

marzoul, 2025-03-13:
Just tested, this may be of interest for other users affected by the present issue :
The [yosys-slang](https://github.com/povik/yosys-slang) project can also bypass the built-in Yosys parser and frontend.
It does not trigger the scalability issue.
```
yosys -l yosys.log -m slang -p "read_slang top.v --top top ; synth_xilinx -arch xc7 -top top ; write_json top.json"
```

marzoul, 2025-03-15:
Correction : the scalability issue is triggered by both the slang plugin (see also #4934) and the ghdl plugin. These frontends just trigger the issue for larger memory size.

IMHO the real issue is that to represent the init values, yosys creates (well it is created in the Yosys RTLIL, in all frontends) one component and memory port, per mem init assignment, hence one **per address to be initialized**, then these things are later removed. This is terribly inefficient.

Maybe this is because it is one of the only solutions to support the fact that the initial blocks in verilog can contain relatively arbitrary code, within which the memory is used as temporary storage, read and written several times, before final init values are reached ? This could be the explanation explain the notion of port "priority" that I see in the Yosys code.

See in genrtlil.cc line 1979, case AST_MEMINIT : here the cell for mem init is created, for each mem init assignment.

marzoul, 2025-03-15:
Yosys frontend, code location:
https://github.com/YosysHQ/yosys/blob/main/frontends/ast/genrtlil.cc#L1979
GHDL plugin, code location:
https://github.com/ghdl/ghdl-yosys-plugin/blob/master/src/ghdl.cc#L479
Slang plugin, code location:
https://github.com/povik/yosys-slang/blob/master/src/slang_frontend.cc#L2882

To my eyes, this way of basically setting a parameter to what should be a vector of init values, is terribly inefficient.
Hence my guess that is was initially done that way as a relatively easy handling of arbitrary init code, at the cost of extra burden on netlist simplification/optimization passes.

povik, 2025-03-15:
> IMHO the real issue is that to represent the init values, yosys creates (well it is created in the Yosys RTLIL, in all frontends) one component and memory port, per mem init assignment, hence one per address to be initialized, then these things are later removed. This is terribly inefficient.

That's not an RTLIL universal -- it depends on the frontend.

> To my eyes, this way of basically setting a parameter to what should be a vector of init values, is terribly inefficient.

Can you elaborate? At least the slang code you are linking should be generating a single $meminit_v2 cell per initialized memory.

marzoul, 2025-03-16:
Oh you're right, sorry. So while I was under gdb I had a look at where most of the time is spent while executing SLANG frontend.
Result : it is spent in call to `convert_const()` at https://github.com/povik/yosys-slang/blob/master/src/slang_frontend.cc#L2880
Certainly proper profiling would be better. I won't be able to dig further in the short term...