YosysHQ/yosys #4610: Yosys Synthesis Fails with std::out_of_range Error in OPT_DEMORGAN Pass

LoSyTe, 2024-09-24:
### Version

yosys 0.41+126

### On which OS did this happen?

Linux

### Reproduction Steps

Hello,

I encountered a crash issue while using Yosys to synthesize a Verilog file. The specific details are as follows:

When running Yosys on the attached Verilog file (rtl.v), the synthesis process crashes during the OPT_DEMORGAN pass. The error message indicates a std::out_of_range exception, which occurs when the tool tries to inspect a $reduce_or cell and push inverters through reduction.

To rule out issues with the file itself, I have checked the Verilog file for syntax errors but did not find any obvious problems. Additionally, I am using the following version of Yosys:Yosys 0.41+126 (git sha1 855ac285f49, g++ 11.4.0-1ubuntu1~22.04 -fPIC -Os)

Here is the synthesis command I used:
yosys -p "
read_verilog rtl.v
hierarchy; proc; opt_clean; opt_dff; opt_demorgan; opt_lut_ins; opt_merge; opt_reduce; opt_dff; opt_expr; opt_muxtree; opt_share; opt_dff; opt_clean; opt_lut_ins; opt_reduce; opt_demorgan;
write_verilog syn_yosys.v"

Here is a snippet of the error log:
[image]
Attached is the Verilog file (rtl.v) that triggers this issue. I hope to get the community's help and attention to resolve this bug.
[yosys_project.zip](https://github.com/user-attachments/files/17108935/yosys_project.zip)

### Expected Behavior

Yosys completes the synthesis without throwing any exceptions.

### Actual Behavior

Yosys with a std::out_of_range error during the OPT_DEMORGAN pass.

Ravenslofty, 2024-09-24:
`opt_lut_ins` doesn't do anything here, they can be removed.

LoSyTe, 2024-09-24:
> `opt_lut_ins` doesn't do anything here, they can be removed.

Thank you for your reply! I have removed the opt_lut_ins pass, but the issue still persists. The current pass sequence is:hierarchy; proc; opt_clean; opt_dff; opt_demorgan; opt_merge; opt_reduce; opt_dff; opt_expr; opt_muxtree; opt_share; opt_dff; opt_clean; opt_reduce; opt_demorgan;

georgerennie, 2024-09-24:
```
read_ilang <<EOT
autoidx 1
module \top
  wire output 1 \Y
  cell $reduce_or $reduce_or$rtl.v:29$20
    parameter \A_SIGNED 0
    parameter \A_WIDTH 0
    parameter \Y_WIDTH 1
    connect \A { }
    connect \Y \Y
  end
end
EOT

opt_demorgan
```

Looks like an issue where `A_WIDTH` is 0

georgerennie, 2024-09-25:
The other part of this is that the rtl contains zero width cells now, which isn't particularly nice. In this case this is happening because `opt_reduce` is being called on `|4'b0000`. It would probably be good to special case this in `opt_reduce` to produce constant 1/0 for these cases

PR #4612 (2024-11-20): opt_demorgan: skip zero width cells
Fixes #4610
closed 2024-11-20 by widlarizer