YosysHQ/yosys #4618: Synthesis fails on MacOS but succeeds on Linux (possibly ABC error)

agrif, 2024-09-29:
### Version

Yosys 0.45+139 (git sha1 8e1e2b9a3, x86_64-apple-darwin21.4-clang++ 14.0.0-1ubuntu1.1 -fPIC -O3)

### On which OS did this happen?

macOS

### Reproduction Steps

Extract [tmpfile-bug.zip](https://github.com/user-attachments/files/17178725/tmpfile-bug.zip) and then run `yosys top.ys`.

### Expected Behavior

Successful synthesis. Although this fails on MacOS, it succeeds on Debian, using the same yosys commit (8e1e2b9a3). Both the mac and linux builds were [downloaded from here](https://github.com/YosysHQ/oss-cad-suite-build/releases/tag/2024-09-29).

### Actual Behavior

Synthesis failure. [The build log](https://github.com/user-attachments/files/17178732/top.txt) ends with:

~~~~
ABC: ** cmd error: aborting 'source <abc-temp-dir>/abc.script'
ABC: The command has to terminate. Boxes are not in a topological order.
ABC: The following information may help debugging (numbers are 0-based):
ABC: Input 0 of BoxA 242 (1stCI = 2079; 1stCO = 425) has TFI with CI 2274,
ABC: which corresponds to output 0 of BoxB 340 (1stCI = 2274; 1stCO = 814).
ABC: In a correct topological order, BoxB should precede BoxA.
ABC: Error: Abc_CommandAbc9If(): Mapping of GIA has failed.

2.23.17.6. Executing AIGER frontend.
ERROR: Can't open input file `/var/folders/5h/w8yjmsfs73x_yzh2cv054svc0000gn/T/yosys-abc-gJsx4M/output.aig' for reading: No such file or directory
~~~~

I think the missing file error is a consequence of the ABC error just before it.

On linux, ABC does not complain about topological ordering, although it does say: `ABC: Mapping (K=8)  :  lut =   2990  edge =   13153  lev =   15 (2.59)  Boxes are not in a topological order. Switching to level computation without boxes.` I'm not sure why ABC would behave differently on linux.

Any change I make to the design, even very minor changes, causes this issue to disappear.

donn, 2024-10-04:
Yosys is in general not cross-OS deterministic -- I've put a couple hours into trying to dissect my synthesis scripts to try to figure out where it starts deviating but it was starting to be something of a timesink and I had to stop.

I can confirm it is per-OS deterministic -- behavior is identical on `aarch64-linux` and `x86_64-linux` for example, and `aarch64-darwin` and `x86_64-darwin`. But the Linux and Darwin ones deviate in sufficiently large designs.

agrif, 2024-10-04:
That's good to know. I noticed that synthesis on Linux and MacOS sometimes have wildly different resource usage, with one sometimes using 3 or 4 times the logic of the other. I've put that down to the Gowin synthesis being a work in progress.

However, failing synthesis entirely on one OS but not another strikes me as a more serious bug.