YosysHQ/yosys #4622: abc segfaults in certain usecase

MrHighVoltage, 2024-10-01:
### Version

Yosys 0.45 (git sha1 9ed031ddd, g++ 11.4.0-1ubuntu1~22.04 -Og -fPIC)

### On which OS did this happen?

Linux

### Reproduction Steps

Compile yosys from source, using the 0.45 tag, which builds ABC from the commit [2188bc7](https://github.com/YosysHQ/abc/tree/2188bc71228b0788569d83ad2b7e7b91ca5dcc09)
Run example provided in this issue : iic-jku/IIC-OSIC-TOOLS#75

### Expected Behavior

According to the reported issue for our container, the script works on a different versions.

### Actual Behavior

ABC segfaults, a backtrace is provided as a comment at the issue there iic-jku/IIC-OSIC-TOOLS#75

widlarizer, 2024-10-01:
Reproduced in ubuntu 22.04 docker as well as up to date nixos unstable with the 0.45 release

wjrforcyber, 2024-10-09:
There's an update on boxes support in `&nf` mapper recently might cause this issue, there are several places need to be fixed, if you don't need boxes in this mapper, revert:
```
git revert 111867432cbfc9d24849bf044b044b8bab6c3b1e 0cb945ebcd8051fc2673d8ebe322a5030b9f948e
```