YosysHQ/yosys #4630: Mapping to flip-flops with an enable in dfflibmap

povik, 2024-10-04:
### Feature Description

`dfflibmap` is responsible for mapping to technology flip-flops as described in a Liberty file. This ticket is to add support for mapping to flip-flop types with an enable, which there's no support for as of yet.

yliu-hashed, 2024-10-08:
I had a similar need. What I did is first use `difflegalize` to reduce all ffs into to a compatable subset of dffs, and use `techmap` to map them manually.
```shell
# script.ys
dfflegalize -cell $_DFF_P_ 0 -cell $_DFFE_PP_ 0
techmap -map techmap_dff.v
```
```verilog
// techmap_dff.v
module \$_DFF_P_ (input D, C, output Q);
MY_DFF _TECHMAP_REPLACE_ (.C(C), .D(D), .Q(Q));
endmodule

module \$_DFFE_PP_ (input D, C, E, output Q);
MY_DFFE _TECHMAP_REPLACE_ (.C(C), .D(D), .E(E), .Q(Q));
endmodule
```

This requires a separate `techmap_dff.v` with all your flops. I guess you can create a techmap verilog file for your library, or find some way to extract and generate the techmap file. 

I also took a look at the code of `dfflibmap` a while back. It looked quite hard-coded. Matching is done directly on Liberty AST. Matching flops with enable is too difficult for me to code up, but I think it can be done, since there's clearly a working Liberty frontend driven by `read_liberty`.

povik, 2024-10-08:
Yes, I have written a similar file for sky130hd to estimate the impact of inferring/not inferring the flip-flops. Let me share it on the off chance it's useful to someone.

```
(* techmap_celltype="$_DFFE_PP_" *)
module dffe_rule(D, C, E, Q);
	input D, C, E;
	output Q;
	sky130_fd_sc_hd__edfxtp_1 _TECHMAP_REPLACE_
		(.CLK(C), .DE(E), .D(D), .Q(Q));
endmodule
```

It might need a bit of code to write generic detection but the information in the Liberty file is there:

```
        ff ("IQ","IQ_N") {
            clocked_on : "CLK";
            next_state : "(D&DE) | (IQ&!DE)";
        }
        pin ("Q") {
            function : "IQ";
        }
```

closed 2024-10-10 by povik