YosysHQ/yosys #4634: make synth stuck at synth -top $::env(DESIGN_NAME) -run fine: {*}$synth_args

janboeye, 2024-10-07:
### Version

Yosys 0.45 (git sha1 3e0dc2ff1, clang++ 14.0.0-1ubuntu1 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

pull the my branch of https://github.com/janboeye/OpenROAD-flow-scripts.

execute make synth

### Expected Behavior

finish the synthesis without any error

### Actual Behavior

when I synthesis Xiangshan which is a big riscv cpu, it stuck at the second round yosys synth command
synth -top $::env(DESIGN_NAME) -run fine: {*}$synth_args
at stage
5.3.1. Executing OPT_EXPR pass (perform const folding).

povik, 2024-10-10:
@janboeye Hello, can you post reproduction steps for this second issue?

janboeye, 2024-10-13:
after upgrade to 0.46, the issue is gone.

Thanks

closed 2024-10-13 by janboeye