YosysHQ/yosys #4663: Add simulation model for Xilinx BRAM

sifferman, 2024-10-13:
### Feature Description

Hello.

My RTL's post-synthesis simulation was failing regression, and I tracked it down to an issue with the BRAM. I just realized that the Xilinx RAMB modules in [techlibs/xilinx/cells_sim.v](https://github.com/YosysHQ/yosys/blob/main/techlibs/xilinx/cells_sim.v) are not implemented for simulation.

Ideally, [techlibs/xilinx/cells_sim.v](https://github.com/YosysHQ/yosys/blob/main/techlibs/xilinx/cells_sim.v) should implement a simulation model for all BRAMs, which can basically be copied from Xilinx's Unisim Library (under Apache 2.0).

* <https://github.com/Xilinx/XilinxUnisimLibrary/blob/master/verilog/src/unisims/RAMB18E1.v>
* <https://github.com/Xilinx/XilinxUnisimLibrary/blob/master/verilog/src/unisims/RAMB18E2.v>
* <https://github.com/Xilinx/XilinxUnisimLibrary/blob/master/verilog/src/unisims/RAMB36E1.v>
* <https://github.com/Xilinx/XilinxUnisimLibrary/blob/master/verilog/src/unisims/RAMB36E2.v>

However, Xilinx's models can't be used directly because they are incompatible with Icarus and Verilator.

Any thoughts on this? I would be happy to do a pull if it would be helpful.

marzoul, 2025-03-31:
Would these simulation models fit the needs ?
https://github.com/fredrequin/verilator_xilinx/
I have absolutely no idea about the quality of these. At least the code is crystal clear and clean :-)
No RAM36BE1 though.

sifferman, 2025-07-26:
I was just able to resolve this issue. I can make a PR at some point, but for now, anyone is welcome to observe my fork here: <https://github.com/sifferman/verilator_xilinx>. (I'll probably clean it up later).

closed 2025-07-26 by sifferman