YosysHQ/yosys #4664: Assert `flow.wire_comb_defs[it].size() == 1'

rroohhh, 2024-10-14:
### Version

Yosys 0.45 (git sha1 9ed031ddd588442f22be13ce608547a5809b62f0, g++ 13.3.0 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

running the following file:
```
read_ilang <<EOF
# Generated by Yosys 0.45 (git sha1 9ed031ddd588442f22be13ce608547a5809b62f0, g++ 13.3.0 -fPIC -O3)
module \arq_delay_67_9159cb8bcee7fcb95582f140960cdae72788d326
  wire width 67 $auto$ghdl.cc:806:import_module$61
  wire width 67 output 1 \data_out
  wire width 277 \internal_bus
  cell $mux \288
    parameter \WIDTH 67
    connect \A 67'x
    connect \B \internal_bus [68:2]
    connect \S 1'x
    connect \Y $auto$ghdl.cc:806:import_module$61
  end
  connect { \internal_bus [137:71] \internal_bus [69] } { $auto$ghdl.cc:806:import_module$61 \internal_bus [0] }
  connect \data_out \internal_bus [137:71]
end
module \arq_tinypacket_64_8_511_8_127_31_4_2_1_6e60e651604b38dd5cb988cd0c5774aa9ac59bc4
  wire width 67 $auto$ghdl.cc:806:import_module$38
  wire width 450 \internal_bus
  wire width 75 output 1 \tx_data
  cell \arq_delay_67_9159cb8bcee7fcb95582f140960cdae72788d326 \tx_delay
    connect \data_out $auto$ghdl.cc:806:import_module$38
  end
  connect \internal_bus [382:316] $auto$ghdl.cc:806:import_module$38
  connect \tx_data [74:8] \internal_bus [382:316]
end
module \arq_wrap
  wire width 75 $auto$ghdl.cc:806:import_module$32
  wire width 181 \internal_bus
  wire width 75 output 1 \tx_data
  cell \arq_tinypacket_64_8_511_8_127_31_4_2_1_6e60e651604b38dd5cb988cd0c5774aa9ac59bc4 \wrapped
    connect \tx_data $auto$ghdl.cc:806:import_module$32
  end
  connect { \internal_bus [175] \internal_bus [157:83] \internal_bus [75:9] } { 1'z $auto$ghdl.cc:806:import_module$32 \internal_bus [157:91] }
  connect \tx_data \internal_bus [75:1]
end
attribute \top 1
module \top
  wire width 9 $0\tx_link_data[0][8:0]
  wire width 75 $indirect$\tx_data$top.v:230$906
  wire width 75 \tx_link_data[0]
  cell \arq_wrap \genblk1[0].arq
    connect \tx_data $indirect$\tx_data$top.v:230$906
  end
  process $proc$top.v:0$967
    sync always
      update \tx_link_data[0] [8:0] $0\tx_link_data[0][8:0]
  end
  connect \tx_link_data[0] [7:0] $indirect$\tx_data$top.v:230$906 [74:67]
end
EOF
write_cxxrtl
```

### Expected Behavior

`write_cxxrtl` runs without triggering a assertion.

### Actual Behavior

The following assertion is triggered:
```
ERROR: Assert `flow.wire_comb_defs[it].size() == 1' failed in backends/cxxrtl/cxxrtl_backend.cc:3269.
```

whitequark, 2024-10-14:
Looks like some of the connections are made in a way I haven't anticipated. I do not currently have time to investigate this issue further.

rroohhh, 2024-10-18:
As a workaround, `flatten; splitcells`  before `write_cxxrtl` works for now.