YosysHQ/yosys #4686: Yosys lacks a warning for a port being driven by 0.

jiangshuirou, 2024-10-28:
### Version

Yosys 0.39+165

### On which OS did this happen?

Linux

### Reproduction Steps

Hello, we performed synthesis on an rtl.v file and noticed that Yosys’s warning messages lacked information about “port y” being driven by 0. For a detailed description, please refer to the question file in the attachment. We kindly ask you to address and resolve this issue promptly.
[Relevant_documents.zip](https://github.com/user-attachments/files/17536637/Relevant_documents.zip)

### Expected Behavior

Yosys should generate a rich number and variety of warning messages without missing any types of warnings. In the attached test case, for the synthesis of "port y", the following 6 warning messages should be present.
1、design top has port y[458] driven by constant 0
2、design top has port y[457] driven by constant 0
3、design top has port y[456] driven by constant 0
4、design top has port y[455] driven by constant 0
5、design top has port y[454] driven by constant 0
6、design top has port y[453] driven by constant 0

### Actual Behavior

The Yoysy warning messages did not include the above-expected 6 warning messages, and the types of warning messages were singular.

rowanG077, 2024-10-28:
Why do you feel an output port being driven to a constant is wrong? This is a totally valid option to me, and doesn't warrant a warning.

QuantamHD, 2024-10-28:
I think I have to agree with @rowanG077 here. This feels like this should be caught by a test bench or a linting tool of some kind. 

A compiler warning for a totally valid state seems like it will just create log spam.

widlarizer, 2024-10-29:
To oversimplify, the goal of Yosys is to take valid Verilog and synthesize valid, usually optimized, physically implementable logic or formal model to verify the design against manual assertions. Outside of formal verification, it hasn't been designed as a tool to look for Verilog constructs that are commonly misused, and it would be difficult to adapt it to that use case

closed 2024-10-29 by widlarizer