YosysHQ/yosys #4687: Yosys synthesize command fails with "Signal `\A' with invalid width range -1" error in cells_map.v.

1353369570, 2024-10-28:
### Version

Yosys 0.46+124 (git sha1 d1695ad9980466ca49d7931ffbc600517682785f, g++ 9.4.0-1ubuntu1~20.04.3 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

read_verilog rtl.v ; synth_xilinx -arch xc7 -top modinst71 -widemux 5

### Expected Behavior

The synthesis should complete without errors, generating the expected netlist for the specified Xilinx architecture

### Actual Behavior

Yosys has the error"/usr/local/bin/../share/yosys/xilinx/cells_map.v:151: ERROR: Signal `\A' with invalid width range -1!"
[image]
[rtl.txt](https://github.com/user-attachments/files/17541263/rtl.txt)

georgerennie, 2024-10-29:
I believe this is the same as #4590, but I don't know enough about the xilinx cells to fix it myself

KrystalDelusion, 2025-03-27:
It even has the same `/* hdl/nnawaq_axi3.vhd:1568:57  */`...

closed 2025-03-27 by KrystalDelusion