YosysHQ/yosys #4693: Yosys seems to be inconsistent with the original design.

smlz123, 2024-10-31:
### Version

0.46+135

### On which OS did this happen?

Linux

### Reproduction Steps

My Verilog original design is as follows：

`module top
(y, clk, wire3, wire2, wire1, wire0);
  output wire [1:0] y;
  input wire [(1'h0):(1'h0)] clk;
  input wire signed [(4'hf):(1'h0)] wire3;
  input wire signed [(3'h6):(1'h0)] wire2;
  input wire [(5'h15):(1'h0)] wire1;
  input wire [(4'hd):(1'h0)] wire0;
  wire signed [(2'h3):(1'h0)] wire4;
  assign y = {wire4, (1'h0)};
  assign wire4 = $signed($signed($signed("") ));
endmodule`

The content of the testbench file is as follows：

``include "syn_yosys.v"
module testbench  ;
  wire [1:0] y;
  reg [(1'h0):(1'h0)] clk;
  reg [(5'h11):(1'h0)] wire3;
  reg signed [(4'he):(1'h0)] wire2;
  reg [(3'h5):(1'h0)] wire1;
  reg signed [(4'hd):(1'h0)] wire0;
  top top1 (.y(y), .clk(clk), .wire3(wire3), .wire2(wire2), .wire1(wire1), .wire0(wire0));
  initial
    begin
      clk = (1'h0);
      {wire3, wire2, wire1, wire0} = (1'h0);
      #10 {wire3, wire2, wire1, wire0} = (251'h597c39e4cef45a644b106c6562b302391f7533b4814ed94f5a3cec8789dfa8f);
      #10 {wire3, wire2, wire1, wire0} = (255'h75aea6bcebd9abb950c280722cde67241fa27f9082ac7ff165be947e4b2b4749);
      #10 {wire3, wire2, wire1, wire0} = (256'hed1f613ee72ab9c2320b7a686681583d01a3aff91b6d657541c74dd192cd5e98);
      #10 {wire3, wire2, wire1, wire0} = (256'hc8b61838e04ed8977bf133c84dc5047719bc8976683dbb23a1c638250970cac9);
      #10 {wire3, wire2, wire1, wire0} = (256'hd047c8dc70e005f95fbf517cb2a1f02bd246a66fe2d1851c6f1f3dd89c004991);
      #10 {wire3, wire2, wire1, wire0} = (256'h9d0993c3dc541dd020807476a99a4d164aaddd7990ba47800233d07752db3984);
      #10 {wire3, wire2, wire1, wire0} = (256'hdf55175761b48b3a36f7bd383a5f737faa7941341e98c5a507407e8be1ada33f);
      #10 {wire3, wire2, wire1, wire0} = (256'h9bbedcb50417f6be124729dc4f6ab5b73d77b033518683daef150934ab04f5cb);
      #10 {wire3, wire2, wire1, wire0} = (256'h84353c58672673a7027ae6c3d7b43a256f053c11dba8fe339e076c560438b1c5);
      #10 {wire3, wire2, wire1, wire0} = (255'h648407cb41b6233830bbe8f77d4254f3207409539d08448dc82458ac61947ec3);
      #10 {wire3, wire2, wire1, wire0} = (256'he1c071a48c91ab9ab1783d53265da7fc878cb879744f311225c0d7cecd2aaea3);
      #10 {wire3, wire2, wire1, wire0} = (255'h7778be7d84431bc37ddd2feb3f742cf599f64c68480eb2e01980b85b4d3defac);
      #10 {wire3, wire2, wire1, wire0} = (247'h59cf8c297d4442d10b1bfaa04d24bc33de0d926ccc886cfbcdf29391d38a70);
      #10 {wire3, wire2, wire1, wire0} = (256'hb8f4344609e482715e3e81863af6b3e63ac76b1698f47cc4f5727dbf4f9a12a3);
      #10 {wire3, wire2, wire1, wire0} = (256'h8cea256c7708318bd2887e695ba1ea5674d6f5e4ebdd8782b8b28da1c16d5eba);
      #10 {wire3, wire2, wire1, wire0} = (252'habeb2156b814891ccda4dd410f1d2dfc3070c0ecc81209bfc4f648341b6e240);
      #10 {wire3, wire2, wire1, wire0} = (255'h4619e49259a6e14e872065d983eec8398a952c272d5dd4cc2b88df64d70bba3b);
      #10 {wire3, wire2, wire1, wire0} = (256'heb3629a9b266222ebdc7f43e715121831757a119f818aab2ba04f0e19fcae59c);
      #10 {wire3, wire2, wire1, wire0} = (255'h67bbd2cbff590e31561a509429df1e6fa45ec3e93775fd94f533d1c2089de7a4);
      #10 {wire3, wire2, wire1, wire0} = (256'ha0c27d59a73b6248467b39e61ff5746d3b8e8aa9173d5ee43db7794293ed0115);
      #10 {wire3, wire2, wire1, wire0} = (256'hc0199281db9e425e422214e075a349dd2bcb83c46e9aacdb630bd5093a8c3a40);
      #10 $finish;
    end
  always
    #5 clk = (~clk);
  always
    @(posedge clk) $strobe ("%b", y);
endmodule
`

### Expected Behavior

The simulation before and after synthesis is consistent.

### Actual Behavior

Inconsistent output before and after synthesis!

georgerennie, 2024-10-31:
I get the same results simulating this before and after synthesis using iverilog, assigning the result 0.

This seems in keeping with the LRM. from 1800-2023:
> 11.10.3 Empty string literal handling
> The empty string literal ("") shall be considered equivalent to the ASCII NUL ("\0"), which has a value
zero (0), which is different from a string "0".

The sign extension of `0` is `0`, thus wire4 should take the value `0` which it seems to

smlz123, 2024-11-01:
[synthesis_4693.zip](https://github.com/user-attachments/files/17599773/synthesis_4693.zip)
These are the files I synthesized. There are indeed inconsistencies before and after synthesis.

georgerennie, 2024-11-01:
I suspect this is an issue with whatever simulator you are using for comparison, not yosys. Your syn_yosys.v in that zip shows that the output `y` is driven by constant `0`, which I believe to be the correct synthesis for the reasons I explained above. Again, `iverilog` provides the same result for both before and after synthesis.

If `file2.txt` is what you expect as the output, I don't see how that could possibly be the result for that input verilog, as the value assigned to `y` is constant in the input verilog whereas `file2.txt` shows changing output.

georgerennie, 2024-11-01:
Maybe your issue is that in your identity testbench you include `rtl.v` from a hardcoded path and might be including the wrong `rtl.v`? Either way if you are going to submit fuzzer generated bugs please ensure they actually reproduce, it has been frustrating that despite members of your group submitting bugs for a while now people have to consistently have to ask for them to be small, reproducible and well explained.

georgerennie, 2024-11-01:
The file `syn_yosys.v` is included in `yosys_testbench.v` by an include statement. This means that it will always be included relative to where you run `iverilog` from. Therefor in your first invocation of the command you are running it with `/doc/zh_zh/test/syn_yosys.v`, and in the second with `/doc/zh_zh/syn_yosys.v`. You can see that the warnings iverilog generates are different because you are compiling different source files.

This is not a Yosys issue.

whitequark, 2024-11-01:
> it has been frustrating that despite members of your group submitting bugs for a while now people have to consistently have to ask for them to be small, reproducible and well explained.

Yosys should have a policy for fuzzer generated bug reports where they are summarily closed if they're not immediately actionable.

closed 2024-11-01 by Ravenslofty