YosysHQ/yosys #4695: Yosys seems to be  inconsistent  with the original design.

smlz123, 2024-10-31:
### Version

0.46+135

### On which OS did this happen?

Linux

### Reproduction Steps

My Verilog original design is as follows：

`module top
(y, clk, wire3, wire2, wire1, wire0);
  output wire [(32'h126):(32'h0)] y;
  input wire [(1'h0):(1'h0)] clk;
  input wire signed [(4'hb):(1'h0)] wire3;
  input wire signed [(4'hc):(1'h0)] wire2;
  input wire [(2'h2):(1'h0)] wire1;
  input wire [(4'he):(1'h0)] wire0;
  wire signed [(4'hf):(1'h0)] wire8;
  reg [(5'h11):(1'h0)] reg5 = (1'h0);
   assign y = {
                 wire8,
                 (1'h0)};
assign wire8 = (wire0[(4'h8):(3'h4)] ? {"x3T4ID"} : reg5[(4'he):(2'h3)]);
endmodule`

The content of the testbench file is as follows：

``include "syn_yosys.v"

module testbench  ;
  wire [(32'h126):(32'h0)] y;
  reg [(1'h0):(1'h0)] clk;
  reg signed [(4'hb):(1'h0)] wire3;
  reg signed [(4'hc):(1'h0)] wire2;
  reg [(2'h2):(1'h0)] wire1;
  reg [(4'he):(1'h0)] wire0;
  top top1 (.y(y), .clk(clk), .wire3(wire3), .wire2(wire2), .wire1(wire1), .wire0(wire0));
  initial
    begin
      clk = (1'h0);
      {wire3, wire2, wire1, wire0} = (1'h0);
      #10 {wire3, wire2, wire1, wire0} = (254'h2ebaed0bc6299c28858ee365d06dfaa0e2690b27d6175a01c377ab98263e3e8f);
      #10 {wire3, wire2, wire1, wire0} = (254'h2373dc17df544faed315c752055b5e25fff0255612bfff4a9ecf0073ba45a902);
      #10 {wire3, wire2, wire1, wire0} = (256'hfd38ef7c3a679a17102d1350b5c735b6722e2f9b3d6b8de28d54e617d5728739);
      #10 {wire3, wire2, wire1, wire0} = (253'h179d4fafd27adb17872425a9bec44b793c17ba91c6476d8222f8a481dc6d6a01);
      #10 {wire3, wire2, wire1, wire0} = (256'h92c15901c895649ce2992cfb7230648dfe2fee6320dafb2c8b55703311c8ca5c);
      #10 {wire3, wire2, wire1, wire0} = (254'h3bdf52d269939856eb57b122e7ccffffa17c129972fa2d125654afbd37ad50f8);
      #10 {wire3, wire2, wire1, wire0} = (256'hccfeadaec0f53592c7c93fdac9b0465567b01ff6e321020ad75dc4fd4d141bd1);
      #10 {wire3, wire2, wire1, wire0} = (256'heee4e46fc790d0ce6ca635c8562f9545ed0db2d761bf5b8f4c96c3ccf7d2629a);
      #10 {wire3, wire2, wire1, wire0} = (255'h65ca2a0b85c260a767e3e25b22e79bd10c709ddb91cb7402a6f8fb45c4715725);
      #10 {wire3, wire2, wire1, wire0} = (255'h56019a59810e7810a6e00d436b113842798f736aac0b5d5b45123e18d88025a9);
      #10 {wire3, wire2, wire1, wire0} = (253'h112b030afda81ac7f40d579a4f0a15c197a24f6f1dfe213283a317708c7b2cc3);
      #10 {wire3, wire2, wire1, wire0} = (256'haadf442f467fe426232f8f873a083aeb3bc51cd53ef072bac0198c82b9597737);
      #10 {wire3, wire2, wire1, wire0} = (254'h2aa2b4ad115215c7a7ea72b9eb119a54976706c280943342d8441a9a270323cb);
      #10 {wire3, wire2, wire1, wire0} = (256'hcd506de2393cf7937bf80abf2b3fd1c5d332590347cfcbaa848e07caa062178c);
      #10 {wire3, wire2, wire1, wire0} = (252'hb9c05f56d8a40958c1de347f7a3a7f5085383431472e5e40bd9166199a14277);
      #10 {wire3, wire2, wire1, wire0} = (255'h6b799d09a82835d12146eb2ff0b0ac5295d043211eb92675d93d9f30c225da89);
      #10 {wire3, wire2, wire1, wire0} = (256'hf6fa630dcc3402f7c17d5d42f5a317193141b5bb6d5cb6421c3383ee615fe21d);
      #10 {wire3, wire2, wire1, wire0} = (256'hc085005c42f89c3f931443c59f527bdcf302a69cb0049c9d44bcf58ccfec59d0);
      #10 {wire3, wire2, wire1, wire0} = (256'h814a670a73d881ca541843b8926d2c87f4067e83614b44e3589c10c9de7b7907);
      #10 {wire3, wire2, wire1, wire0} = (256'h90865987e57a88afd0473be45350811861e357a740e78dfcfc9f911765a3a391);
      #10 {wire3, wire2, wire1, wire0} = (256'hd169a4cb4fd75a8fab14c5e9d7b6fa068d20f31c9493093eed60236a725e187a);
      #10 $finish;
    end
  always
    #5 clk = (~clk);
  always
    @(posedge clk) $strobe ("%b", y);
endmodule
`

### Expected Behavior

The simulation before and after synthesis is consistent.

### Actual Behavior

Inconsistent output before and after synthesis.

georgerennie, 2024-10-31:
I see the same behaviour before and after synthesis using iverilog. The least significant bits of the output are:
```
0000000000000000
1001001010001000
1001001010001000
1001001010001000
0000000000000000
1001001010001000
1001001010001000
1001001010001000
1001001010001000
1001001010001000
1001001010001000
1001001010001000
1001001010001000
1001001010001000
1001001010001000
1001001010001000
1001001010001000
1001001010001000
1001001010001000
1001001010001000
1001001010001000
1001001010001000
```

smlz123, 2024-11-01:
[synthesis_3.zip](https://github.com/user-attachments/files/17599689/synthesis_3.zip)
There are the files I synthesized.The simulation before and after synthesis is inconsistent.

smlz123, 2024-11-01:
[synthesis_4695.zip](https://github.com/user-attachments/files/17599787/synthesis_4695.zip)
These are the files I synthesized. There are indeed inconsistencies before and after synthesis.

georgerennie, 2024-11-01:
My comments from https://github.com/YosysHQ/yosys/issues/4693#issuecomment-2451847268 also apply here. I am unable to reproduce the differences you see and get the same values before and after synthesis when simulated with iverilog. I cannot reproduce the `file2.txt` you show.

Again the results in your `file2.txt` are impossible for the verilog you show. In your testcase, y is assigned `{wire8, 1'h0}`. `wire8` is only 16 bits wide so only the 17 LSBs of `y` could possibly have a non-zero value yet `file2.txt` shows non-zero values in many other places.

If you still believe there is a bug here, please provide a small, self-contained input file showing missynthesis with an explanation of what you expect the output to be, rather than a testbench that is hard to debug due to the random values. Such a testbench is fine for fuzzing, but it is not helpful for those trying to help triage your issue - particularly when I am unable to reproduce your testbench results.

KrystalDelusion, 2025-03-27:
Closing this is invalid.

closed 2025-03-27 by KrystalDelusion