YosysHQ/yosys #4696: Yosys seems to be inconsistent with the original design.

smlz123, 2024-10-31:
### Version

0.46+135

### On which OS did this happen?

Linux

### Reproduction Steps

My Verilog original design is as follows：

`module top
(y, clk, wire3, wire2, wire1, wire0);
  output wire [(32'h1cf):(32'h0)] y;
  input wire [(1'h0):(1'h0)] clk;
  input wire [(4'ha):(1'h0)] wire3;
  input wire [(4'hc):(1'h0)] wire2;
  input wire signed [(5'h12):(1'h0)] wire1;
  input wire [(3'h4):(1'h0)] wire0;
  wire signed [(4'hb):(1'h0)] wire48;
  reg [(4'h8):(1'h0)] reg32 = (1'h0);
  wire [(3'h4):(1'h0)] wire10;
  wire signed [(3'h4):(1'h0)] wire7;
  reg [(3'h6):(1'h0)] reg5 = (1'h0);
  reg signed [(3'h7):(1'h0)] reg4 = (1'h0);
  assign y = {wire48,        
                 wire10,   
                 wire7,
                 reg5,
                 reg4,
                 (1'h0)};
  assign wire7 = (-reg5);
  assign wire10 = $signed(($unsigned("") >>> reg4[(1'h0):(1'h0)]));
  assign wire48 = reg32;
endmodule`

The content of the testbench file is as follows：

``include "syn_yosys.v"
module testbench  ;
  wire [(32'h1cf):(32'h0)] y;
  reg [(1'h0):(1'h0)] clk;
  reg [(4'ha):(1'h0)] wire3;
  reg [(4'hc):(1'h0)] wire2;
  reg signed [(5'h12):(1'h0)] wire1;
  reg [(3'h4):(1'h0)] wire0;
  top top1 (.y(y), .clk(clk), .wire3(wire3), .wire2(wire2), .wire1(wire1), .wire0(wire0));
  initial
    begin
      clk = (1'h0);
      {wire3, wire2, wire1, wire0} = (1'h0);
      #10 {wire3, wire2, wire1, wire0} = (255'h7d186e0d15991474bf1b776bb67adf4a0008afa6cac3f56a2db06ae8cab7e411);
      #10 {wire3, wire2, wire1, wire0} = (255'h7410d1e50133946433bf656cd353b98e4e867fb9dfab3e205c31307eef2639f0);
      #10 {wire3, wire2, wire1, wire0} = (254'h27f88cdd44577fef60e63ff297647e4a4367d929806da8bcdf72be13db86a71c);
      #10 {wire3, wire2, wire1, wire0} = (253'h16bef8f3ea19c50ae858f3a69ee803c45512737c1d91b67dc23321f9526efdb4);
      #10 {wire3, wire2, wire1, wire0} = (253'h171769618a70faac84df993fd750e4b73604f4f77a712fbbf2036baeed2a15f9);
      #10 {wire3, wire2, wire1, wire0} = (256'h82160d6ef1e2b644fa08d0a3ff7dce3e799cc82e487c02a59e9cc3c37f5ed821);
      #10 {wire3, wire2, wire1, wire0} = (255'h4d53fa11d0864d991aae2185fb66c3b8dc9fd43df14fa50e90d6c5db94efae7b);
      #10 {wire3, wire2, wire1, wire0} = (256'hd801ad6b3be108b1276b186708caa00c9f9f8d3ad2492fabe826275e3ee1bbbd);
      #10 {wire3, wire2, wire1, wire0} = (252'hb0e5a9a7f614c4f2927abc56344f331d1f5736c60c5154d9481db2eef40d58e);
      #10 {wire3, wire2, wire1, wire0} = (256'hcd37062e634b7b7398580c6520073480c69e1c4d3bd9179c90a2461d28b2d7fe);
      #10 {wire3, wire2, wire1, wire0} = (255'h56f214dab11e4cb9f3b611fb532305fff12daafbfcd6a5b4bf56989218c80d1b);
      #10 {wire3, wire2, wire1, wire0} = (255'h7e3a90a352c86eb58e7f5625f0138eec346c0f9532545b2a2e1cc1dff159a5e6);
      #10 {wire3, wire2, wire1, wire0} = (255'h58389b1237cd448ebd818af032bb72cbdd1690ff227bbd75463cb26025af9e5b);
      #10 {wire3, wire2, wire1, wire0} = (256'ha4a531d240fea89516438c56b70b2f655100e85ada7be7f87e39b4a5ff0f8f12);
      #10 {wire3, wire2, wire1, wire0} = (253'h1f18faa33beffd5d276ab968043c93e038caed9b08b02b83c939e7a0081bbbaf);
      #10 {wire3, wire2, wire1, wire0} = (256'haa9a55d97e151206a1d2b3016f045354b36fcb4b61797132a1fa3d6ef0e4c7f9);
      #10 {wire3, wire2, wire1, wire0} = (256'hda19b544379e7bf7fbb1f4c07d423e68a92472777e72f3ab58176a6d9b7def36);
      #10 {wire3, wire2, wire1, wire0} = (256'hb02964d54c9cbc14f7705aa76046b85a23ae88fb6bd3c8e29a376caae2a049e0);
      #10 {wire3, wire2, wire1, wire0} = (256'hc0139e6030cc3a12f2ec9ddcc9a4c376f44d4d892b772607e62dc98810e9539e);
      #10 {wire3, wire2, wire1, wire0} = (255'h501bca538d10daad3190a79f5152d3b827aa4b7675c950eef2ba95750518ea55);
      #10 {wire3, wire2, wire1, wire0} = (256'hc24a19ff3cb64f43d8768a8dfbe660dcd057c533325da2d185f8d6f645ad003a);
      #10 $finish;
    end
  always
    #5 clk = (~clk);
  always
    @(posedge clk) $strobe ("%b", y);
endmodule
`

### Expected Behavior

The simulation before and after synthesis is consistent.

### Actual Behavior

Inconsistent output before and after synthesis.

povik, 2024-10-31:
When you say before and after synthesis please report the commands you are running.

WenNarin, 2024-10-31:
Please be more professional.

whitequark, 2024-10-31:
There wasn't anything unprofessional in @povik's response.

yyiloe, 2024-10-31:
This is a beginner，it is meaningless

WenNarin, 2024-10-31:
> There wasn't anything unprofessional in @povik's response.

My point is that the question raised in @smlz123  does not provide a proper bug report.

smlz123, 2024-10-31:
> When you say before and after synthesis please report the commands you are running.

Commands used for synthesis:
```
yosys
read_verilog rtl.v
synth
write_verilog syn_yosys.v
```
then,you can simulate them by the above testbench,
please note that `include"syn_yosys"` should changed to `include"rtl.v"` when you simulate the rtl.v.
Looking forward to your reply.

georgerennie, 2024-10-31:
Unlike the other bugs you opened, I can reproduce this but its still very much in the realm of very contrived fuzzer generated bugs given it relies on `""` to create a zero width wire.

In the future it helps greatly if you can minimize bugs into a small, human readable testcase that doesn't need a testbench but instead can be demonstrated by showing the inequivalence between the representation before and after synthesis with an explanation as to why you believe the result is from incorrect synthesis.

I can manually reduce it to this
```verilog
module top(input wire a, output wire [3:0] b);
assign b = $signed("" >> a);
endmodule
```

which results in the following after just `read_verilog`
```
autoidx 2

attribute \cells_not_processed 1
attribute \src "bug.v:1.1-3.10"
module \top

  attribute \src "bug.v:2.21-2.27"
  wire width 0 $shr$bug.v:2$1_Y

  attribute \src "bug.v:1.44-1.45"
  wire width 4 output 2 \b

  attribute \src "bug.v:1.23-1.24"
  wire input 1 \a

  attribute \src "bug.v:2.21-2.27"
  cell $shr $shr$bug.v:2$1
    parameter \Y_WIDTH 0
    parameter \B_WIDTH 1
    parameter \A_WIDTH 0
    parameter \B_SIGNED 0
    parameter \A_SIGNED 0
    connect \Y { }
    connect \B \a
    connect \A { }
  end

  connect \b 4'x
end
```

Looks like it may be an issue in the verilog frontend to do with how optimization of `$signed` works if given a 0 width non-constant value

georgerennie, 2024-10-31:
This is because sign extension of signed variables defaults to `Sx` if the width is `0` in rtlil.cc https://github.com/YosysHQ/yosys/blob/8fb73e18ffac5d6c24b97273bebfbd3a086320c9/kernel/rtlil.cc#L4979-L4983

smlz123, 2024-11-01:
> This is because sign extension of signed variables defaults to `Sx` if the width is `0` in rtlil.cc
> 
> https://github.com/YosysHQ/yosys/blob/8fb73e18ffac5d6c24b97273bebfbd3a086320c9/kernel/rtlil.cc#L4979-L4983
> 
> I'm not sure if that needs to be the case - semantically I think you can make the argument that the sign extension of a zero width signal should be constant zero - but maybe there is some use for this behaviour i'm not thinking about. Maybe the better solution would be to special case zero width strings and give them a width?

Assuming we have a more complex design where the sign extension of a zero-width signal, denoted as Sx, is meaningful. For example:
```
module shift_example #(parameter WIDTH = 0) (
    input wire signed [WIDTH-1:0] in,
    input wire [3:0] shift,
    output wire signed [WIDTH-1:0] out
);
    assign out = in >>> shift;
endmodule
```
In this module, the WIDTH parameter can be any value, including zero. If WIDTH is zero, then the widths of both the in and out signals are zero. 
In this case, the logic of sign extension becomes very important.

```
module complex_design #(parameter WIDTH = 0) (
    input wire signed [WIDTH-1:0] data,
    input wire [3:0] control,
    output wire signed [WIDTH-1:0] result
);
    always @(*) begin
        if (control == 4'b0000) begin
            result = data >>> 1;
        end else begin
            result = data <<< 1;
        end
    end
endmodule
```
Also, in this design, the control signal determines the shift direction of the data signal. If WIDTH is zero and the sign extension is zero, then the result will always be zero, which might mask potential errors in the design. However, if the sign extension is Sx, it can expose issues in the design earlier, allowing designers to identify and correct these problems in a timely manner.

georgerennie, 2024-11-01:
Those examples you provide will not trigger this behaviour, as you cannot declare zero width signals like that. They will expand to `[-1:0]` which is interpreted as an ascending bit range vector (in keeping with other tools like iverilog and verilator) - see examples in 6.9.1 of IEEE 1800-2023.

I suspect it is only possible to create 0 width signals in the verilog frontend in a small number of corner case situations like the empty string used above. Another way you could do it is
```verilog
module top(input wire a, output wire [3:0] b);
assign b = $signed({0{1'b1}} >> a);
endmodule
```
which triggers the same behaviour.

My preference is probably towards making `extend_u0` sign extend zero width signals with 0, or maybe outright rejecting width 0 signals in the frontend (but again im not sure if there is some legitimate use they may come in, I could see the repetition operator being used with 0 repetitions under some corner case from parameters).

georgerennie, 2024-11-01:
FWIW the LRM rightfully forbids this pattern with the replication, 11.4.12.1 in IEEE 1800-2023

> A replication operation may have a multiplier with a value of zero. This is useful in parameterized code. A
replication with a zero multiplier is considered to have a size of zero and is ignored. Such a replication shall
appear only within a concatenation in which at least one of the operands of the concatenation has a positive
size.

Verilator correctly catches this in `--lint-only` mode, which is highly recommended if using the yosys verilog frontend as the frontend is known (and accepted) to be overly accepting. Thus I don't think this case is necessarily an issue we care about, although the empty string is as the LRM allows empty strings.

whitequark, 2024-11-01:
> FWIW the LRM rightfully forbids this pattern with the replication, 11.4.12.1 in IEEE 1800-2023

The problem with completely forbidding this is that RTLIL does allow zero width signals and I think we don't have any way other than `{0{1'b0}}` to emit them in the Verilog backend. (I originally thought that `""` does it, but Verilog strings are null-terminated, so `""` is a 8-bit value.)

georgerennie, 2024-11-01:
Yeah I think its fine to just leave `{0{1'b0}}` for now given verilator linting would catch it.

> I originally thought that "" does it, but Verilog strings are null-terminated, so "" is a 8-bit value.

I believe they are not null-terminated (e.g. `"A"` would be `8'h65` not `16'h6500`), it is just that the LRM explicitly declares zero-length strings to have a null value.

> 11.10.3 Empty string literal handling
> The empty string literal ("") shall be considered equivalent to the ASCII NUL ("\0"), which has a value
zero (0), which is different from a string "0".

I guess to implement this we can just special case convert empty string literals to `"\0"`

georgerennie, 2024-11-01:
Page 113 of IEEE 1800-2023 gives the following (which seems to agree with things from previous LRM versions).
```verilog
typedef logic [15:0] r_t;
r_t r;
string b = "";

r = {"H",""}; // yields "H\0". "" is converted to 8'b0
b = {"H",""}; // yields "H". "" is the empty string
```

The interpretation seems to be contextual based on whether it assigned to a `string` or not, but I believe yosys doesn't support `string` so just special casing `""` as `8'b0` should be fine.

whitequark, 2024-11-01:
> I believe they are not null-terminated (e.g. `"A"` would be `8'h65` not `16'h6500`),

Oh right, they're big-endian too. I completely missed that then...

PR #4697 (2024-11-04): frontends/ast.cc: special-case zero width strings as "\0"
* Fixes #4696 

Special cases zero width strings as `"\0"` as required by IEEE 1364-2005 5.2.3.3
> The null string ( "") shall be considered equivalent to the ASCII NUL ("\0"), which has a value zero (0),
which is different from a string "0".
commit dbfca1bd (2024-11-04): frontends/ast.cc: special-case zero width strings as "\0"

* Fixes #4696
closed 2024-11-04 by Ravenslofty