YosysHQ/yosys #4708: Yosys does not accept module port lists with .port_identifier

ldoolitt, 2024-11-05:
### Version

Yosys 0.46 (git sha1 e97731b9d, g++ 12.2.0-14 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

```
cat > same_port.v << EOT
module same_port (.a(i), .b(i)); // Name 'i' is declared inside the
                    // module as a inout port. Names 'a' and 'b' are
                    // defined for port connections.
inout i;
wire i;
endmodule
EOT
yosys -p "read_verilog same_port.v"
```

### Expected Behavior

No crash.
IEEE Std 1354-2001 Version C gives the following BNF:
```
port ::=
  [ port expression ]
  | .port_identifier([port_expression])
```
and the same_port.v example file above is straight from the standard, p.187,
with the last three lines to complete the picture.

Vivado and iverilog both handle this case correctly.

### Actual Behavior

```

 /----------------------------------------------------------------------------\
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |  Copyright (C) 2012 - 2024  Claire Xenia Wolf <claire@yosyshq.com>         |
 |  Distributed under an ISC-like license, type "license" to see terms        |
 \----------------------------------------------------------------------------/
 Yosys 0.46 (git sha1 e97731b9d, g++ 12.2.0-14 -fPIC -O3)

-- Running command `read_verilog same_port.v' --

1. Executing Verilog-2005 frontend: same_port.v
Parsing Verilog input from `same_port.v' to AST representation.
same_port.v:1: ERROR: syntax error, unexpected TOK_ID, expecting '.'
```

ldoolitt, 2024-11-05:
Verilator also fails this test.
See its issue 5588 https://github.com/verilator/verilator/issues/5588

KrystalDelusion, 2025-03-27:
I can confirm this is accepted with the Verific frontend giving RTLIL:
```
autoidx 1

attribute \library "work"
attribute \hdlname "same_port"
attribute \src "same_port.v:1.8-1.17"
module \same_port

  attribute \src "same_port.v:1.26-1.31"
  wire inout 2 \b

  attribute \src "same_port.v:1.19-1.24"
  wire inout 1 \a

  connect \a \b
end
```
and verilog
```verilog
(* \library  = "work" *)
(* hdlname = "same_port" *)
(* src = "same_port.v:1.8-1.17" *)
module same_port(a, b);
  (* src = "same_port.v:1.19-1.24" *)
  inout a;
  wire a;
  (* src = "same_port.v:1.26-1.31" *)
  inout b;
  wire b;
  assign a = b;
endmodule
```
and reading back the verilog output with `read_verilog` works fine.  I'm not convinced the semantics are strictly the same, but given #3426 exists, directionality of assigns seems to be a loose concept.