YosysHQ/yosys #4712: handling of register arrays broken after bdb5d45591d7501825349bedcb2952d0b80a1112

gadfort, 2024-11-06:
### Version

Yosys 0.44+60 (git sha1 bdb5d4559, g++ 13.2.0-23ubuntu4 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

[sc_issue_simon_cipher_job0_floorplan0_20241106-014731.tar.gz](https://github.com/user-attachments/files/17651881/sc_issue_simon_cipher_job0_floorplan0_20241106-014731.tar.gz)

```
tar xvf sc_issue_simon_cipher_job0_floorplan0_20241106-014731.tar.gz
cd sc_issue_simon_cipher_job0_floorplan0_20241106-014731
./run.sh
```

From: https://github.com/siliconcompiler/siliconcompiler/issues/2965

### Expected Behavior

Correctly handling the register array (as before bdb5d45591d7501825349bedcb2952d0b80a1112).

### Actual Behavior

Before bdb5d45591d7501825349bedcb2952d0b80a1112 this netlist handled `reg [31:0] key_schedule [0:43];` correctly, afterwards the `proc` call in `synth` appears to be incomplete.
```
yosys> opt_clean

4.5. Executing OPT_CLEAN pass (remove unused cells and wires).
Warning: Ignoring module simon_cipher because it contains processes (run 'proc' command first).
```
the resultant netlist ends with a bunch of unmapped instantances
```

            "$_ALDFFE_PPP_": 32,
            "$_AND_": 438,
            "$_MUX_": 2554,
            "$_NOT_": 414,
            "$_OR_": 524,
            "$_XOR_": 296,
            "$scopeinfo": 2,
```

georgerennie, 2024-11-06:
I'm happy to have a look at this (seeing as its my commit that introduced the issue), but i'm struggling to reproduce things with your files. Could you give a reproducer that doesn't require openlane etc and is just a yosys invocation on the right files, as well as a description about exactly what you see in the output with the new version that you don't see with the old one (I'm not quite sure what the behaviour you want for "Correctly handling the register array" is).

georgerennie, 2024-11-06:
I found the sources by searching around on github and have minimized it into a small testcase:

```
read_rtlil <<EOT
autoidx 1
module \top
  wire input 1 \clk
  wire input 2 \rst

  wire input 3 \a_r
  wire input 4 \a_n
  wire input 5 \b_n

  wire \a
  wire \b

  process $proc
    sync high \rst
      update \a \a_r
      update \b \b
    sync posedge \clk
      update \a \a_n
      update \b \b_n
  end
end
EOT

proc_dff
stat
```

With old yosys we get
```
2. Executing PROC_DFF pass (convert process syncs to FFs).
Creating register for signal `\top.\a' using process `\top.$proc'.
Warning: Async reset value `\a_r' is not constant!
  created $aldff cell `$procdff$1' with positive edge clock and positive level non-const reset.
Creating register for signal `\top.\b' using process `\top.$proc'.
  created $dff cell `$procdff$4' with positive edge clock.
  
   Number of wires:                  8
   Number of wire bits:              8
   Number of public wires:           7
   Number of public wire bits:       7
   Number of ports:                  5
   Number of port bits:              5
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              1
   Number of cells:                  3
     $aldff                          1
     $dff                            1
     $mux                            1
```

with new we get
```
2. Executing PROC_DFF pass (convert process syncs to FFs).
Creating register for signal `\top.\a' using process `\top.$proc'.
Warning: Async reset value `\a_r' is not constant!
  created $aldff cell `$procdff$3' with positive edge clock and positive level non-const reset.

=== top ===

   Number of wires:                  8
   Number of wire bits:              8
   Number of public wires:           7
   Number of public wire bits:       7
   Number of ports:                  5
   Number of port bits:              5
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              1
   Number of cells:                  2
     $aldff                          1
     $reduce_or                      1
```

It seems to not be handling the `b` case in the process - will look into it

georgerennie, 2024-11-06:
I see what I had done, I had mistakenly used `return` to finish processing a rule when actually it should use `continue` to go to the next rule because they are being processed in a loop. Will put up a patch and a testcase to catch this.

gadfort, 2024-11-07:
@georgerennie I'm sorry I accidentally uploaded the wrong testcase. I just tested the fix in the PR and it does correctly synthesize the testcase I meant to uploaded.

closed 2024-11-20 by povik

dkongg, 2024-11-22:
Hi guys, the synthesis seems to not be working in Yosys 0.47 now

georgerennie, 2024-11-22:
We need more information than that to do anything about it. We never actually received a testcase for this issue so i can't evaluate it.

Are you setting the same failure in this thread? if so please post the yosys version you are using (the git commit hash in particular). If not please open a new issue with details.

dkongg, 2024-11-22:
Yes it is the same issue. Yosys version and verilog code below:

Yosys 0.47 (git sha1 647d61dd9, g++ 9.4.0-1ubuntu1~20.04.2 -fPIC -O3)
[simon_cipher.zip](https://github.com/user-attachments/files/17866529/simon_cipher.zip)

 This was working before on Yosys 0.44

georgerennie, 2024-11-22:
Our bug fix (#4714) was not included in version 0.47 and is not present in commit 647d61d. As you can see above, it was only merged two days ago. Please use a build of yosys off of the up to date main branch.

PR #4714 (2024-11-20): proc_dff: fix early return bug
Fixes #4712 which is a bug introduced in #4569, where multiple rules in processes were not being considered if an asynchronous one was encountered as the code erroneously returned from the function instead of continuing the processing loop.