YosysHQ/yosys #4722: muxcover generates x states

miradarya, 2024-11-08:
### Version

0.47

### On which OS did this happen?

Linux, Windows

### Reproduction Steps

I haven't been able to verify the fix for #3591 for some time now. Unfortunately however, commit 822c7b0341064c3d0d9b142dc6ce61c66797159a breaks muxcover, at least for some non-trivial cases. I currently have a design where muxcover after 822c7b0341064c3d0d9b142dc6ce61c66797159a generates x states on the S input of several $_MUX8_ cells. There are no x states present in the design before executing muxcover. Reverting muxcover.cc to before 822c7b0341064c3d0d9b142dc6ce61c66797159a does not generate x states (but still has the cost estimation issue mentioned in #3591). At this point, I cannot provide the actual design that reproduces the issue. I am unsure though, how to create a minimal example.

### Expected Behavior

design with correctly connected $_MUX8_ cells

### Actual Behavior

design with x states on (at least) S input of $_MUX8_ cells

KrystalDelusion, 2025-03-27:
There is work in progress documentation for using the `bugpoint` command to minimize failing designs: https://yosyshq.readthedocs.io/projects/yosys/en/docs-preview-bugpoint/using_yosys/bugpoint.html#minimizing-rtlil-designs-with-bugpoint.  Can you have a go at following those instructions to provide a minimized example?  Let me know if there is anything that is unclear and I can try to help.

marzoul, 2025-03-31:
This seems similar to what I observed in https://github.com/YosysHQ/yosys/issues/4590
Maybe the reproducer can be of use here.

KrystalDelusion, 2025-03-31:
@marzoul actually your example in #4590 is fully defined both before and after the `muxcover`, so I don't think this is related.

miradarya, 2025-04-01:
Thanks for pointing me to the `bugpoint` command. I haven't got a minimized example yet, but working on it.