YosysHQ/yosys #4747: Inconsistent simulation before and after yosys synthesis(1)

iagrigorov, 2024-11-15:
### Version

Yosys 0.44 (git sha1 80ba43d, g++ 11.4.0-2ubuntu1~20.04 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

1. Install the necessary dependencies (skip this step if you have already installed `verilator` and `zip`):
```
sudo apt-get install zip verilator
```
2. Download the attached `.zip` archive. Then change to the directory where this archive is located and unpack it:
```
unzip files.zip
```
3. Go to `files` directory:
```
cd files
```
4. Launch `yosys` like this:
```
yosys -p"read_verilog crp.v sbox1.v sbox2.v sbox3.v sbox4.v sbox5.v sbox6.v sbox7.v sbox8.v key_sel3.v des3.v; proc; opt -nodffe; memory; pmuxtree; flatten -scopename; opt -nodffe -fast; aigmap; write_verilog des3_dut.v"
```
5. Change the top module name of the synthesized module:
```
sed -i 's/module des3/module des3_dut/g' des3_dut.v
```
6. Launch `verilator` like this:
```
verilator crp.v sbox1.v sbox2.v sbox3.v sbox4.v sbox5.v sbox6.v sbox7.v sbox8.v key_sel3.v des3.v des3_dut.v des3_testbench.sv -Wno-fatal +1364-2005ext+v +1800-2017ext+sv --binary --top-module des3_testbench
```
7. Launch the simulation:
```
./obj_dir/Vdes3_testbench
```
[files.zip](https://github.com/user-attachments/files/17774318/files.zip)

### Expected Behavior

The expected behavior is the successful simulation result (without any discrepancies between the source model and the synthesized model).

### Actual Behavior

`verilator` gives the following output:
```
 \desIn_gold =           2037806020
 \key1_gold =         186539420
 \key2_gold = 72057592410395906
 \key3_gold =        1220731108
 \decrypt_gold = 0
 \roundSel_gold = 60
 \clk_gold = 1
DUT Error at time                   10
 \desOut :
Expected value  5573564902638284270, Got Value  1816601759550881946
[1500] %Fatal: des3_testbench.sv:100: Assertion failed in TOP.des3_testbench
%Error: des3_testbench.sv:100: Verilog $stop
Aborting...
Aborted (core dumped)
```

jix, 2024-11-15:
See my response to #4748

closed 2024-11-15 by jix

iagrigorov, 2024-11-15:
This not an arbitrary verilog code. This a code from benchmark `iwls05`.
Almost every design in this benchmark passes. You can check it yourself. And almost every design in this benchmark contains timing annotations. For example, there is no such error on almost exactly the same design:
[files.zip](https://github.com/user-attachments/files/17775525/files.zip)