YosysHQ/yosys #4766: Error: Abc_CommandAbc9If(): Mapping of GIA has failed.

spth, 2024-11-24:
### Version

0.47 (also 0.40)

### On which OS did this happen?

Linux

### Reproduction Steps

Compile and install yosys 0.40 or yosys 0.47 (I did not try other versions) from the release tarball on Debian GNU/Linux testing on aarch64. I was not able to reproduce the issue using yosys 0.47 (I did not try other versions) on 
Debian GNU/Linux testing on amd64.

Try to compile the attached Verilog source via `yosys -l icesynth2.yosyslog -p "read_verilog -sv cpu2_nosv.v; synth_ice40 -top cpu -json icesynth2.json"`

[cpu2_nosv.v.gz](https://github.com/user-attachments/files/17894340/cpu2_nosv.v.gz)

### Expected Behavior

The source gets compiled.

### Actual Behavior

I get an error message:

~~~~
[…]
2.41.16.5. Executing ABC9.
Running ABC command: "<yosys-exe-dir>/yosys-abc" -s -f <abc-temp-dir>/abc.script 2>&1
ABC: ABC command line: "source <abc-temp-dir>/abc.script".
ABC: 
ABC: + read_lut <abc-temp-dir>/input.lut 
ABC: + read_box <abc-temp-dir>/input.box 
ABC: + &read <abc-temp-dir>/input.xaig 
ABC: + &ps 
ABC: <abc-temp-dir>/input : i/o =    134/    238  and =   13939  lev =  214 (10.84)  mem = 0.28 MB  box = 1129  bb = 145
ABC: + &scorr 
ABC: Warning: The network is combinational.
ABC: + &sweep 
ABC: + &dc2 
ABC: + &dch -f 
ABC: + &ps 
ABC: <abc-temp-dir>/input : i/o =    134/    238  and =   14626  lev =   60 (4.57)  mem = 0.27 MB  ch = 2028  box = 971  bb = 145
ABC: + &if -W 250 -v 
ABC: ** cmd error: aborting 'source <abc-temp-dir>/abc.script'
ABC: The command has to terminate. Boxes are not in a topological order.
ABC: The following information may help debugging (numbers are 0-based):
ABC: Input 0 of BoxA 379 (1stCI = 872; 1stCO = 1815) has TFI with CI 1005,
ABC: which corresponds to output 0 of BoxB 452 (1stCI = 1005; 1stCO = 2128).
ABC: In a correct topological order, BoxB should precede BoxA.
ABC: Error: Abc_CommandAbc9If(): Mapping of GIA has failed.

2.41.16.6. Executing AIGER frontend.
ERROR: Can't open input file `/tmp/yosys-abc-CVKznF/output.aig' for reading: No such file or directory
~~~~

povik, 2024-11-24:
Given the below this is probably a Yosys bug (not an ABC bug) in that we have prepared the XAIGER input wrong.

```
ABC: The command has to terminate. Boxes are not in a topological order.
ABC: The following information may help debugging (numbers are 0-based):
ABC: Input 0 of BoxA 379 (1stCI = 872; 1stCO = 1815) has TFI with CI 1005,
ABC: which corresponds to output 0 of BoxB 452 (1stCI = 1005; 1stCO = 2128).
ABC: In a correct topological order, BoxB should precede BoxA.
```

KrystalDelusion, 2025-03-27:
Is this still happening in the latest version of Yosys?

closed 2025-03-28 by KrystalDelusion