YosysHQ/yosys #4774: ECP5 LUT6 enhancement

peepo, 2024-11-27:
### Feature Description

Lattice ECP5 software has support for LUT6, would be helpful to me and possibly others if this was available in open source variant.

see also: https://github.com/YosysHQ/prjtrellis/issues/240

Ravenslofty, 2024-11-28:
again, you can use LUT4s and PFUMX/L2MUX6 to build a LUT6, if you need to.

but I guess another question is: what do you need direct LUT6 instantiation for?

peepo, 2024-11-28:
Hi Lofty,
Please could you elaborate, as this approach does not seem well documented?
Apologies "LUT4s and PFUMX/L2MUX6 to build a LUT6, if you need to." means absolutely nothing to me, and a search for PFUMX/L2MUX6 has no relevant links in my Google results.
many thanks

Ravenslofty, 2024-11-28:
The ECP5 does not have LUT6s in hardware; it has LUT4s, which you can multiplex together to form LUT6s. To make a LUT5, you combine two LUT4s with a PFUMX, with the fifth input as the PFUMX selector. To make a LUT-N (N > 5), you combine two LUT-(N-1)s with an L6MUX21.

peepo, 2024-11-28:
Sorry to ask but, would it be possible to provide the code for an example?
I'm not finding an example in the wild, and have no idea what you intend, in terms of verilog.
Instantiating in Diamond is a single line of code, relatively easy to understand.
For various reasons, I am trying to move code base back to open source, however LUT6 is a dealbreaker on this project.
My preferred learning style is 'cookbook', as you can probably tell!!
many thanks!

Ravenslofty, 2024-11-28:
Different approach: if you know the LUT mask, you can right-shift it by the LUT inputs (concatenated).

jix, 2024-11-28:
a cookbook version of @Ravenslofty's last suggestion would be:
```
module LUT6(output O, input I0, I1, I2, I3, I4, I5);
  parameter [63:0] INIT = 0;
  assign O = INIT >> {I5, I4, I3, I2, I1, I0};
endmodule
```

peepo, 2024-11-28:
Jannis, appreciated! however without wishing to appear lazy, could you please include LUT4, as else....

peepo, 2024-11-28:
The elephant in the bath?
Please excuse my ignorance, but isn't this approach more suited to ice40 where LUT6 is not available?
ECP5 has LUT6, so is there evidence that this technique will use LUT6 efficiently?
It seems to me that Diamond offers the capability to ensure 1:1 compatibility.
In anycase, this is not intended as a 'wind-up', so if I'm mistaken, please accept my apologies.

widlarizer, 2024-11-28:
ECP5 has no LUT6 blocks available in hardware. If the proprietary software pretends otherwise, it's composing them from LUT4.

> Since there are two LUT4s per slice, a LUT5 can be constructed within one slice. Larger look-up tables such as LUT6, LUT7, and LUT8 can be constructed by concatenating other slices

from Lattice data sheet [FPGA-DS-02012-2.4](https://www.mouser.com/datasheet/2/225/FPGA_DS_02012_2_4_ECP5_ECP5G_Family_Data_Sheet-1022822.pdf) section 2.2.2.1. The surrounding descriptions and diagrams are helpful as well.

Without being more concrete of what you'd like to achieve and how, there's no clear steps for us to take to help. Jannis has shown Verilog code implementing a LUT6. Synthesis will map it to whatever LUTs are available in hardware. Are you trying to instantiate individual LUTs manually?

peepo, 2024-11-28:
ok got it, thanks all!
At the time I'll give it a whirl, should be fine...

peepo, 2024-11-28:
arghh: how to set a 6bit mask?....

widlarizer, 2024-11-28:
Anyhow, I believe we've concluded discussing the actual declared topic of this issue

closed 2024-11-28 by widlarizer