YosysHQ/yosys #4780: Verilog: Mixing integer and real values causes error

flafflar, 2024-11-28:
### Version

Yosys 0.47+135 (git sha1 6f3376cbe, g++ 14.2.1 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

A Verilog code that can cause the bug is the following:
```verilog
module test (
    input [31:0] a,
    output q
);

parameter VALUE = 100;

assign q = a > VALUE * 1.2;

endmodule
```

### Expected Behavior

Yosys should parse that Verilog code into RTLIL successfully

### Actual Behavior

The `read_verilog` command fails with the following error :
```
ERROR: Found error in internal cell \test.$gt$test.v:8$1 ($gt) at kernel/rtlil.cc:1333:
  attribute \src "test.v:8.12-8.27"
  cell $gt $gt$test.v:8$1
    parameter \Y_WIDTH 1
    parameter \B_WIDTH 32
    parameter \A_WIDTH 32
    parameter \B_SIGNED 1
    parameter \A_SIGNED 0
    connect \Y $gt$test.v:8$1_Y
    connect \B 120
    connect \A \a
  end
```

KrystalDelusion, 2024-11-28:
The error being raised is actually in an `if (check_matched_sign) {` block, so I suspect the problem is because it's comparing B as signed and A as unsigned that is causing the error.  Notice `connect \B 120`; which means it correctly multiplies `100*1.2`.

flafflar, 2024-11-28:
You're probably right, changing the code to this:
```verilog
module test (
    input signed [31:0] a,
    output q
);

parameter VALUE = 100;

assign q = a > VALUE * 1.2;

endmodule
```
throws no error.

closed 2025-03-27 by KrystalDelusion