YosysHQ/yosys #4785: read_verilog: array of instances parsing assertion failure

Muxianesty, 2024-11-29:
### Version

Yosys 0.44 (git sha1 80ba43d26, clang++-18 18.1.8 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

```verilog
module test(in1, in2, out1);
  input in1, in2;
  output out1;

  nand  #2 t_nand[0:7](out1, in1, in2);
  nand  #2 x_nand[0:3](out1, in1, in2), y_nand[4:7](out1, in1, in2);
endmodule
```

### Expected Behavior

> IEEE Std 1364-2005:
>    7. Gate- and switch-level modeling
>      7.1 Gate and switch declaration syntax
>        7.1.5 The range specification
>          An array of instances shall have a continuous range. One instance identifier shall be
>          associated with only one range to declare an array of instances.
>          The range specification shall be optional. If no range specification is given, a single
>          instance shall be created.

The module body specified in reproduction steps was taken directly from the standard (page 78). It should be parsed just fine.

### Actual Behavior

```shell
 /----------------------------------------------------------------------------\
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |  Copyright (C) 2012 - 2024  Claire Xenia Wolf <claire@yosyshq.com>         |
 |  Distributed under an ISC-like license, type "license" to see terms        |
 \----------------------------------------------------------------------------/
 Yosys 0.44 (git sha1 80ba43d26, clang++-18 18.1.8 -fPIC -O3)

yosys> read_verilog module.v 

1. Executing Verilog-2005 frontend: module.v
Parsing Verilog input from `module.v' to AST representation.
Generating RTLIL representation for module `\test'.
ERROR: Assert `new_cell->children.at(0)->type == AST_CELLTYPE' failed in frontends/ast/simplify.cc:2686.
```

Muxianesty, 2024-12-02:
I noticed that if `nand`-wrapper is used, the error disappears.

```verilog
module nand_test(output out1, input in1, input in2);
  nand #2 result(out1, in1, in2);
endmodule

module test(in1, in2, out1);
  input in1, in2;
  output out1;

  nand_test t_nand[0:7](out1, in1, in2);
  nand_test x_nand[0:3](out1, in1, in2), y_nand[4:7](out1, in1, in2);
endmodule
```

KrystalDelusion, 2025-03-24:
So, this *should* be raising the error `Cell arrays of primitives are currently not supported`, which is also why it works when you use the wrapper module.  But it doesn't seem to be getting caught properly.

PR #4959 (2025-07-20): simplify: Skip AST_PRIMITIVE in AST_CELLARRAY
_What are the reasons/motivation for this change?_

Close #4785.

We have an input error about cell arrays of primitives not being supported, but the `AST_PRIMITIVE` is simplified before it can be checked, leading to an assertion error instead.

_Explain how this is achieved._

Don't simplify `AST_PRIMITIVE` in `AST_CELLARRAY`, similar to #3467.

_If applicable, please suggest to reviewers how they can test the change._

Run `yosys tests/verilog/bug4785.ys` with/without this patch.
closed 2025-07-20 by KrystalDelusion