YosysHQ/yosys #4786: `cxxrtl` output port changes value after back to back `.step()`

rroohhh, 2024-11-29:
### Version

Yosys 0.45 (git sha1 9ed031ddd588442f22be13ce608547a5809b62f0, g++ 13.3.0 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

running the following script
```
read_rtlil <<EOF
module \arq_delay
  wire $auto$bugpoint.cc:258:simplify_something$28
  wire $auto$ghdl.cc:806:import_module$5
  wire $auto$ghdl.cc:806:import_module$6
  wire $auto$ghdl.cc:806:import_module$7
  wire $auto$ghdl.cc:806:import_module$8
  wire input 1 \arq_valid
  wire output 4 \client_next
  wire \client_valid
  wire input 5 \clock
  wire width 8 input 2 \data_in
  wire width 8 output 3 \data_out
  wire offset 20 \internal_bus[21:20][20]
  wire offset 21 \internal_bus[21:20][21]
  wire offset 1 \internal_bus[9:1][1]
  wire input 6 \reset
  cell $mux \31
    parameter \WIDTH 8
    connect \A 8'x
    connect \B \data_in
    connect \S \client_valid
    connect \Y \data_out
  end
  cell $not \41
    parameter \A_SIGNED 0
    parameter \A_WIDTH 1
    parameter \Y_WIDTH 1
    connect \A \client_valid
    connect \Y $auto$ghdl.cc:806:import_module$5
  end
  cell $not \45
    parameter \A_SIGNED 0
    parameter \A_WIDTH 1
    parameter \Y_WIDTH 1
    connect \A $auto$bugpoint.cc:258:simplify_something$28
    connect \Y $auto$ghdl.cc:806:import_module$6
  end
  cell $and \46
    parameter \A_SIGNED 0
    parameter \A_WIDTH 1
    parameter \B_SIGNED 0
    parameter \B_WIDTH 1
    parameter \Y_WIDTH 1
    connect \A $auto$ghdl.cc:806:import_module$6
    connect \B $auto$ghdl.cc:806:import_module$5
    connect \Y $auto$ghdl.cc:806:import_module$7
  end
  cell $and \48
    parameter \A_SIGNED 0
    parameter \A_WIDTH 1
    parameter \B_SIGNED 0
    parameter \B_WIDTH 1
    parameter \Y_WIDTH 1
    connect \A \arq_valid
    connect \B $auto$ghdl.cc:806:import_module$7
    connect \Y $auto$ghdl.cc:806:import_module$8
  end
  cell $mux \58[0]
    parameter \WIDTH 1
    connect \A 1'x
    connect \B 1'1
    connect \S $auto$ghdl.cc:806:import_module$8
    connect \Y \internal_bus[21:20][20]
  end
  cell $dff \79[0]
    parameter \CLK_POLARITY 1
    parameter \WIDTH 1
    connect \CLK \clock
    connect \D \internal_bus[21:20][20]
    connect \Q \client_valid
  end
  connect \internal_bus[21:20][21] \internal_bus[21:20][20]
  connect \internal_bus[9:1][1] \client_valid
end
EOF

write_cxxrtl out.cxx
exec -expect-return 0 -- c++ -I$(yosys-config --datdir)/include/backends/cxxrtl/runtime main.cpp && ./a.out
```

using a `main.cpp` containing
```c++
#include "cxxrtl/cxxrtl.h"
#include "out.cxx"
#include <cstdio>

auto main() -> int {
  cxxrtl_design::p_arq__delay dut;
  dut.step();

  for(int i = 0; i < 5; i++) {
    dut.p_data__in.set(i);
    dut.p_arq__valid.set(true);

    if (i > 1) {
        dut.p_client__next.set(true);
    }

    dut.p_clock.set(false);
    dut.step();

    dut.p_clock.set(true);
    dut.step();
    auto a = dut.p_data__out.get<int>();
    dut.step();
    auto b = dut.p_data__out.get<int>();
    if (a != b) {
      printf("fail\n");
      return 83;
    }
  }
  return 0;
}
```

You can also find the `yosys` script aswell as the `main.cpp` in the attached [repro.zip](https://github.com/user-attachments/files/17962769/repro.zip).

### Expected Behavior

I expect the `a.out` binary to never print "fail", or expressed differently for the value of a output port to not change between subsequent `.step()` calls.

### Actual Behavior

The value of the output port changes between `.step()` calls.

whitequark, 2024-11-30:
> ```verilog
>   output data_out;
>   wire data_out;
>   assign data_out = client_valid;
>   assign client_valid_next  = ~client_valid;
> ```

Oh. You're taking feedback from your own output. There's a bit of code that special-cases outputs (it was related to reducing the amount of delta cycles if all you have is a model with a bunch of LED outputs), and since normal code rarely does that, it's buggy.

whitequark, 2024-11-30:
Try removing this line and seeing if it helps:

https://github.com/YosysHQ/yosys/blob/f04b89972123325dc48704f30e4f118aa24fca05/backends/cxxrtl/cxxrtl_backend.cc#L3127

rroohhh, 2024-12-02:
> Oh. You're taking feedback from your own output. There's a bit of code that special-cases outputs (it was related to reducing the amount of delta cycles if all you have is a model with a bunch of LED outputs), and since normal code rarely does that, it's buggy.

What do you mean by `feedback from your own output`? This is just a comb output driven from a internal register. I don't see how there is feedback from my output?

> Try removing this line and seeing if it helps:
> 
> https://github.com/YosysHQ/yosys/blob/f04b89972123325dc48704f30e4f118aa24fca05/backends/cxxrtl/cxxrtl_backend.cc#L3127

I tried this and it did not change the output. Also I don't quite see how this could change anything, as `!module->get_bool_attribute(ID::top)` should always be false for a single module (or a flattened hierarchy) and therefore the `if` never triggering anyways, right?

whitequark, 2024-12-02:
> This is just a comb output driven from a internal register. I don't see how there is feedback from my output?

Oh, you're right; I misread the code.