YosysHQ/yosys #4798: Synthesis with -nowidelut gives drastically better results

t-wallet, 2024-12-05:
### Version

0.45+139

### On which OS did this happen?

Linux

### Reproduction Steps

While synthesizing a SHA3 design on the Colorlight 5A-75B (Lattice ECP5 FPGA) board, I noticed that the LUT usage was way higher than when I used Vivado to synthesize the same design on a Xilinx board I own. It turns out that synthesizing with `-nowidelut` drastically reduced resource usage and significantly improved timing as well.

The problem was in the theta step of the algorithm:

```verilog
module sha3theta (
  input  wire[24:0][24:0][63:0] i_state,
  output wire[24:0][24:0][63:0] o_state
);

  wire[5:0][63:0] sum_sheet;

  genvar i;
  generate
  for (i = 0; i < 5; i++) begin
    assign sum_sheet[i] =
      i_state[0][i] ^
      i_state[1][i] ^
      i_state[2][i] ^
      i_state[3][i] ^
      i_state[4][i];
  end
  endgenerate

  genvar row, col;
  generate
  for (row = 0; row < 5; row++) begin
    for (col = 0; col < 5; col++) begin
      assign o_state[row][col] =
        i_state[row][col] ^
        sum_sheet[(col - 1) % 5] ^
        {sum_sheet[(col + 1) % 5][62:0], sum_sheet[(col + 1) % 5][63:63]};
    end
  end
  endgenerate;

endmodule
```

Synthesizing this module with `-nowidelut` gives the following resource usage:
```
Number of cells: 1972
     LUT4 1972
```

And without the flag:
```
Number of cells: 9512
     L6MUX21 1188
     LUT4 5656
     PFUMX 2668
```

### Expected Behavior

I would expect that at least the timing would be improved by the usage of the wide muxes found on the ECP5 FPGA.

### Actual Behavior

Synthesizing the full design with `-nowidelut` improved the timing, `Fmax` being 90 MHz, vs 60 MHz without the flag. Of course I verified that this module was in the critical path.

Ravenslofty, 2024-12-09:
A situation like this occurs when ABC9 has a mismatch between predicted and actual delay.

Without `-nowidelut`, ABC9 predicts 1.225ns of delay, while with `-nowidelut` ABC9 predicts 1.457ns of delay.

For fun, I set the ABC9 delay target to 10ns, and got a solution using 1920 LUT4s.

Out of curiosity, what's your testing methodology here? Did you just run the design once under nextpnr?

madscientist159, 2025-04-28:
I'm seeing the same behavior with our Kestrel project [1] after updaing Yosys to GIT hash `8bdbf797d` from GIT hash `57c9eb70f`

**Without `--yosys-nowidelut`:**

```
19.48. Printing statistics.

=== rcs_arctic_tern_bmc_card ===

   Number of wires:              44689
   Number of wire bits:         242758
   Number of public wires:       44689
   Number of public wire bits:  242758
   Number of ports:                 73
   Number of port bits:            167
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:              68309
     $scopeinfo                     83
     CCU2C                        3111
     CLKDIVF                         1
     DDRDLLA                         1
     DELAYG                         70
     DP16KD                        126
     DQSBUFM                         4
     ECLKBRIDGECS                    1
     ECLKSYNCB                       1
     EHXPLLL                         2
     IDDRX1F                         5
     IDDRX2DQA                      32
     L6MUX21                        72
     LUT4                        39135
     MULT18X18D                     20
     ODDRX1F                         6
     ODDRX2DQA                      36
     ODDRX2DQSB                      4
     ODDRX2F                        27
     OSCG                            1
     PFUMX                        4808
     TRELLIS_DPR16X4               826
     TRELLIS_FF                  19836
     TRELLIS_IO                     64
     TSHX2DQA                       32
     TSHX2DQSA                       4
     USRMCLK                         1

19.49. Executing CHECK pass (checking for obvious problems).
Checking module rcs_arctic_tern_bmc_card...
Found and reported 0 problems.

19.50. Executing JSON backend.

Warnings: 26 unique messages, 42 total
End of script. Logfile hash: 2612a34b88, CPU: user 292.98s system 1.20s, MEM: 2503.12 MB peak
Yosys 0.52+117 (git sha1 8bdbf797d, g++ 12.2.0-14 -fPIC -O3)
Time spent: 14% 1x autoname (48 sec), 13% 38x opt_clean (46 sec), ...
Info: constraining clock net 'cpu_clk' to 63.00 MHz
Info: constraining clock net 'eth_rx_clk' to 125.00 MHz
Info: constraining clock net 'eth_tx_clk' to 125.00 MHz
Info: constraining clock net 'clk125' to 125.00 MHz
Info: constraining clock net 'eth_clocks_rx' to 125.00 MHz

Info: Logic utilisation before packing:
Info:     Total LUT4s:     50313/83640    60%
Info:         logic LUTs:  39135/83640    46%
Info:         carry LUTs:   6222/83640     7%
Info:           RAM LUTs:   3304/10455    31%
Info:          RAMW LUTs:   1652/20910     7%

Info:      Total DFFs:     19836/83640    23%
```
`nextpnr` final timing:
`Warning: Max frequency for clock                          '$glbnet$sdrio_clk': 48.14 MHz (FAIL at 63.12 MHz)`

**With `--yosys-nowidelut`:**

```
19.48. Printing statistics.

=== rcs_arctic_tern_bmc_card ===

   Number of wires:              38778
   Number of wire bits:         238398
   Number of public wires:       38778
   Number of public wire bits:  238398
   Number of ports:                 73
   Number of port bits:            167
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:              63556
     $scopeinfo                     83
     CCU2C                        3095
     CLKDIVF                         1
     DDRDLLA                         1
     DELAYG                         70
     DP16KD                        126
     DQSBUFM                         4
     ECLKBRIDGECS                    1
     ECLKSYNCB                       1
     EHXPLLL                         2
     IDDRX1F                         5
     IDDRX2DQA                      32
     LUT4                        39278
     MULT18X18D                     20
     ODDRX1F                         6
     ODDRX2DQA                      36
     ODDRX2DQSB                      4
     ODDRX2F                        27
     OSCG                            1
     TRELLIS_DPR16X4               826
     TRELLIS_FF                  19836
     TRELLIS_IO                     64
     TSHX2DQA                       32
     TSHX2DQSA                       4
     USRMCLK                         1

19.49. Executing CHECK pass (checking for obvious problems).
Checking module rcs_arctic_tern_bmc_card...
Found and reported 0 problems.

19.50. Executing JSON backend.

Warnings: 26 unique messages, 42 total
End of script. Logfile hash: 5c228157c2, CPU: user 284.82s system 1.12s, MEM: 2168.31 MB peak
Yosys 0.52+117 (git sha1 8bdbf797d, g++ 12.2.0-14 -fPIC -O3)
Time spent: 14% 38x opt_clean (46 sec), 14% 1x memory_libmap (46 sec), ...
Info: constraining clock net 'cpu_clk' to 63.00 MHz
Info: constraining clock net 'eth_rx_clk' to 125.00 MHz
Info: constraining clock net 'eth_tx_clk' to 125.00 MHz
Info: constraining clock net 'clk125' to 125.00 MHz
Info: constraining clock net 'eth_clocks_rx' to 125.00 MHz

Info: Logic utilisation before packing:
Info:     Total LUT4s:     50424/83640    60%
Info:         logic LUTs:  39278/83640    46%
Info:         carry LUTs:   6190/83640     7%
Info:           RAM LUTs:   3304/10455    31%
Info:          RAMW LUTs:   1652/20910     7%

Info:      Total DFFs:     19836/83640    23%
```

`nextpnr` final timing:
`Warning: Max frequency for clock                          '$glbnet$sdrio_clk': 57.80 MHz (FAIL at 63.12 MHz)`

For comparison, Yosys `57c9eb70f` would easily meet timing at  >65MHz with the same design.

[1] https://gitlab.raptorengineering.com/kestrel-collaboration/kestrel-litex/litex-boards