YosysHQ/yosys #4808: Problem with new `tclapi.cc` on CentOS 7

akashlevy, 2024-12-10:
### Version

0.47+211

### On which OS did this happen?

Linux

### Reproduction Steps

Happens on CentOS 7 with TCL installed and enabled. Problem started when `tclapi.cc` was introduced (@povik). Seems that some TCL header is missing that defines `CONST` as `const` on this platform. The problem disappears if we edit `tclapi.cc` to add the define:

```
#ifdef YOSYS_ENABLE_TCL
#define CONST const
#include "tclTomMath.h"
#include "tclTomMathDecls.h"
#endif
```

Happy to make a PR if deemed necessary. The alternative is to include the proper TCL header that has `#define CONST const` in it (I don't know which one that is...)

### Expected Behavior

Compiles without error

### Actual Behavior

```
In file included from /usr/include/tclTomMath.h:18,
                 from kernel/tclapi.cc:25:
/usr/include/tclTomMathDecls.h:327:56: error: ‘CONST’ has not been declared
  327 | EXTERN int              TclBN_mp_read_radix(mp_int *a, CONST char *str,
      |                                                        ^~~~~
/usr/include/tclTomMathDecls.h:327:56: error: two or more data types in declaration of ‘str’
/usr/include/tclTomMathDecls.h:327:71: error: expected ‘)’ before ‘,’ token
  327 | EXTERN int              TclBN_mp_read_radix(mp_int *a, CONST char *str,
      |                                            ~                          ^
      |                                                                       )
/usr/include/tclTomMathDecls.h:328:33: error: expected unqualified-id before ‘int’
  328 |                                 int radix);
      |                                 ^~~
/usr/include/tclTomMathDecls.h:511:44: error: ‘CONST’ has not been declared
  511 |     int (*tclBN_mp_read_radix) (mp_int *a, CONST char *str, int radix); /* 36 */
      |                                            ^~~~~
/usr/include/tclTomMathDecls.h:511:44: error: two or more data types in declaration of ‘str’
/usr/include/tclTomMathDecls.h:511:59: error: expected ‘)’ before ‘,’ token
  511 |     int (*tclBN_mp_read_radix) (mp_int *a, CONST char *str, int radix); /* 36 */
      |                                ~                          ^
      |                                                           )
/usr/include/tclTomMathDecls.h:511:61: error: expected unqualified-id before ‘int’
  511 |     int (*tclBN_mp_read_radix) (mp_int *a, CONST char *str, int radix); /* 36 */
      |                                                             ^~~
make[3]: *** [Makefile:816: kernel/tclapi.o] Error 1
```

povik, 2024-12-10:
It's fortunate you have caught it before we make a release in the coming days.

It looks like it's plain `tcl.h` we are missing to include: https://github.com/tcltk/tcl/blob/c824b6262d5a423f7085fa6e5cfacd98bc98e0b1/generic/tcl.h#L239

Can you confirm adding the include does fix it?

akashlevy, 2024-12-10:
Nice, yeah I thought it might be `tcl.h`. Will try right now and get back to you

akashlevy, 2024-12-10:
Hmm, that did not fix it. It seems that the `tcl.h` on the CentOS 7 platform I'm testing with does not define `CONST`

akashlevy, 2024-12-10:
I'm also including Verific, which comes with its own `tcl.h`. Not sure if this is conflicting somehow. Looking into it

akashlevy, 2024-12-10:
Ok, if you load the TCL includes first and THEN the Yosys internal libraries, it works. My header looks like:
```
#ifdef YOSYS_ENABLE_TCL
#include "tclTomMath.h"
#include "tclTomMathDecls.h"
#endif

#include "kernel/yosys.h"
#include "kernel/rtlil.h"
#include "libs/json11/json11.hpp"
```
I'm not sure exactly why this works, but it seems `yosys.h` is preventing the TCL includes from being processed properly.

akashlevy, 2024-12-10:
Ah. `yosys_common.h` has the following problematic line:
```
#  undef CONST
```
Not sure why that is there, but it is the underlying cause of trouble.

povik, 2024-12-10:
I've found this commit to introduce the defines originally (before it was moved between files): ac988cfac50050314f974dd1c7cc29ba4b46935d

Apparently CXXRTL had an issue with the defines

closed 2024-12-10 by povik

akashlevy, 2024-12-10:
Thank you for the resolution, Martin