YosysHQ/yosys #4816: synth_xilinx error in simple demo

lehaifeng000, 2024-12-13:
### Version

0.47

### On which OS did this happen?

Linux

### Reproduction Steps

[image]

command: `yosys  -p "synth_xilinx -flatten -nowidelut -arch xc7 -top top; json -o synth.json top;" top.v`

top.v

```
module top(
    input wire clk,  
    input key1,
    output led  
);
    FDRE fdre_inst2(
        .C(clk),
        .D(~key1),
        .CE(1'b1),
        .Q(led)
    );

endmodule
```

### Expected Behavior

no

### Actual Behavior

no

lehaifeng000, 2024-12-13:
error message:

2.47. Executing XILINX_DFFOPT pass (optimize FF control signal usage).
Optimizing FFs in top.
terminate called after throwing an instance of 'std::out_of_range'
  what():  dict::at()
[1]    31402 IOT instruction (core dumped)  /home/lhf/src/eda/yosys/yosys -p  top.v

nakengelhardt, 2024-12-13:
The code in `xilinx_dffopt` didn't consider the case of unconnected ports. Until this is fixed, you can avoid running into this bug by assigning something to the `R` port of `FDRE`:
```
module top(
    input wire clk,  
    input key1,
    output led  
);
    FDRE fdre_inst2(
        .C(clk),
        .D(~key1),
        .CE(1'b1),
        .Q(led),
        .R(1'b0)
    );

endmodule
```

lehaifeng000, 2024-12-16:
> The code in `xilinx_dffopt` didn't consider the case of unconnected ports. Until this is fixed, you can avoid running into this bug by assigning something to the `R` port of `FDRE`:
>
> ```
> module [...]

Thanks!