YosysHQ/yosys #4822: Built yosys.wasm is too large

LSTM-Kirigaya, 2024-12-18:
### Version

Yosys 0.48+5 (git sha1 7a362f1f7, clang++ 18.1.2-wasi-sdk -Oz)

### On which OS did this happen?

Linux, WebAssembly

### Reproduction Steps

Disable all the features in Makefile

```makefile
# features (the more the better)
ENABLE_TCL := 0
ENABLE_ABC := 0
ENABLE_GLOB := 0
ENABLE_PLUGINS := 0
ENABLE_READLINE := 0
ENABLE_EDITLINE := 0
ENABLE_GHDL := 0
ENABLE_VERIFIC := 0
ENABLE_VERIFIC_SYSTEMVERILOG := 0
ENABLE_VERIFIC_VHDL := 0
ENABLE_VERIFIC_HIER_TREE := 0
ENABLE_VERIFIC_YOSYSHQ_EXTENSIONS := 0
ENABLE_VERIFIC_EDIF := 0
ENABLE_VERIFIC_LIBERTY := 0
ENABLE_COVER := 0
ENABLE_LIBYOSYS := 0
ENABLE_ZLIB := 0

...

# Needed for environments that can't run executables (i.e. emscripten, wasm)
DISABLE_SPAWN := 1
# Needed for environments that don't have proper thread support (i.e. emscripten, wasm--for now)
DISABLE_ABC_THREADS := 1
```

then run

```
make config-wasi
make -j 48
du yosys.wasm -h
```

output:
```
242M    yosys.wasm
```

The build wasm is too large (previous version of emcc build wasm is only 12MB)

btw, wasi sdk is `wasi-sdk-24.0-x86_64-linux`

### Expected Behavior

Expect generated `yosys.wasm` with size of `<20MB`

### Actual Behavior

Generated `yosys.wasm` with size of `242MB`

whitequark, 2024-12-19:
Do you happen to build with debug information? If you post the output of objdump -h that would help.

LSTM-Kirigaya, 2025-02-15:
Thanks for reply, when I enter `objdump -h yosys.wasm`, output is 

```
objdump: yosys.wasm: file format not recognized
```

So, I use `wasm-objdump` instead, here is the output of `wasm-objdump -h yosys.wasm`

```
yosys.wasm:     file format wasm 0x1
module name: <yosys.wasm>

Sections:

     Type start=0x0000000b end=0x000002e5 (size=0x000002da) count: 78
   Import start=0x000002e8 end=0x0000056f (size=0x00000287) count: 17
 Function start=0x00000573 end=0x00004bc1 (size=0x0000464e) count: 17995
    Table start=0x00004bc3 end=0x00004bca (size=0x00000007) count: 1
   Memory start=0x00004bcc end=0x00004bcf (size=0x00000003) count: 1
   Global start=0x00004bd2 end=0x00004ec4 (size=0x000002f2) count: 108
   Export start=0x00004ec6 end=0x00004ed9 (size=0x00000013) count: 2
     Elem start=0x00004edd end=0x0000953d (size=0x00004660) count: 1
     Code start=0x00009542 end=0x00eb30b1 (size=0x00ea9b6f) count: 17995
     Data start=0x00eb30b5 end=0x00f7ee25 (size=0x000cbd70) count: 2
   Custom start=0x00f7ee2a end=0x031374f4 (size=0x021b86ca) ".debug_loc"
   Custom start=0x031374f8 end=0x03220f35 (size=0x000e9a3d) ".debug_abbrev"
   Custom start=0x03220f3a end=0x09ca796e (size=0x06a86a34) ".debug_info"
   Custom start=0x09ca7973 end=0x0c1c64b7 (size=0x0251eb44) ".debug_str"
   Custom start=0x0c1c64bc end=0x0dea8ced (size=0x01ce2831) ".debug_line"
   Custom start=0x0dea8cf2 end=0x0ed5ffb0 (size=0x00eb72be) ".debug_ranges"
   Custom start=0x0ed5ffb5 end=0x0f10d094 (size=0x003ad0df) "name"
   Custom start=0x0f10d097 end=0x0f10d135 (size=0x0000009e) "producers"
   Custom start=0x0f10d137 end=0x0f10d170 (size=0x00000039) "target_features"
```

whitequark, 2025-02-16:
It looks like the majority of the file size is taken up by the debug information. You should call `strip` on the resulting binary, or pass `-Wl,--strip-all` option to the compiler driver. See https://github.com/YoWASP/yosys/blob/develop/build.sh for an example.

LSTM-Kirigaya, 2025-02-21:
Thanks for the inspiration !
Finally, I use `wasm-strip` in https://github.com/WebAssembly/wabt and use following command to make `yosys.wasm` shrink to just `27MB`:

```bash
wasm-strip yosys.wasm
```

Amazing! Thanks very much :D