YosysHQ/yosys #4826: INIT of LUT1 must be 2 bit ?

lehaifeng000, 2024-12-23:
### Version

0.47

### On which OS did this happen?

Linux

### Reproduction Steps

top.v
```
module top(
    output  led 
);
    LUT1 #(
        .INIT(1'b1)
    ) lut1_inst (
        .I0(1'b1),
        .O(led)
    );

endmodule
```

command: `yosys  -p "synth_xilinx -flatten -nowidelut -arch xc7 -top top; json -o synth.json top;" top.v`

### Expected Behavior

synth pass

### Actual Behavior

2.49. Executing OPT_LUT_INS pass (discard unused LUT inputs).
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 1) >= this->size() (which is 1)

lehaifeng000, 2024-12-23:
if the INIT is set  .INIT(2'b1), it will pass

nakengelhardt, 2025-01-15:
Yes, see the [7 series LUT1 primitive documentation](https://docs.amd.com/r/2024.1-English/ug953-vivado-7series-libraries/LUT1):

[image]