YosysHQ/yosys #4828: Inconsistent simulation before and after synthesis of complex assignment expressions

FSY369, 2024-12-24:
### Version

Yosys 0.34 (git sha1 4a1b55992, gcc 7.5.0-6ubuntu2 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Run the script in the terminal (after removing the .log suffix from the filename)：
        yosys -p "
             read_verilog rtl.v         
             synth
             write_verilog syn_yosys1.v"
        iverilog -o wave -y syn_yosys1.v yosys_testbench.v
        vvp wave > file1.txt
        
        
        iverilog -o wave1 -y rtl.v yosys_testbench1.v
        vvp wave1 > file2.txt
[rtl.v.log](https://github.com/user-attachments/files/18237601/rtl.v.log)
[yosys_testbench.v.log](https://github.com/user-attachments/files/18237602/yosys_testbench.v.log)
[yosys_testbench1.v.log](https://github.com/user-attachments/files/18237603/yosys_testbench1.v.log)

### Expected Behavior

Pre-synthesis re-simulation and post-synthesis simulation should be consistent

### Actual Behavior

Pre-synthesis re-simulation and post-synthesis simulation should be consistent
[image]
[image]

whitequark, 2024-12-24:
You are submitting a bug report produced using fuzzing tools. With such bug reports, it is often hard to tell whether the issue reported is an issue in Yosys or an issue with the test harness.

Unless you submit a minimized example clearly demonstrating the construct that causes the inconsistent behavior your bug report will be closed as invalid.

FSY369, 2024-12-26:
Ok thanks a lot for your suggestion, I have simplified the code and implemented only some reg and wire definition and instantiation modules in the code again and found that the problem is in this code in the top module:
 `always
    @(posedge clk)
            begin
              reg259 <= (($signed((wire250[(5'h11):(5'h10)] ?
                      (~&reg251) : $signed(wire237))) - $signed(($unsigned(wire234) ?
                      (wire3 * reg248) : (wire246 ~^ reg251)))) ?
                  $unsigned((wire242 >>> (~{wire243, wire237}))) : wire238);                     
            end `
It was found to cause inconsistencies in the simulation function before and after synthesis
[rtl1.txt](https://github.com/user-attachments/files/18250496/rtl1.txt)

whitequark, 2024-12-28:
Please submit another complete RTL file with only the offending construct in it. It should be only a few lines long.

FSY369, 2024-12-31:
[rtl2.txt](https://github.com/user-attachments/files/18279091/rtl2.txt)

whitequark, 2024-12-31:
This is an improvement but it still contains redundant code that does not appear to be core to the issue. For example, `param280` and `param281` are never used inside of the body of the module, and are clearly redundant.

Please submit a complete RTL file without every construct that is not relevant to the issue at hand.

georgerennie, 2024-12-31:
See https://github.com/YosysHQ/yosys/issues/997 for an example of a well reported bug found by fuzzing. Note that it includes the following

- A short human readable testcase that contains nothing but the constructs needed to trigger the bug
- A hypothesis about what constructs Yosys is treating incorrectly and what the expected behaviour should be
- The testcase is small enough to be put in a small markdown block in the github issue instead of requiring a download. This generally makes it easier for people to look at the issue and see if its something they are likely to be able to fix. Having a big testcase that requires a separate download reduces the amount of people that will bother to look at your issue

whitequark, 2024-12-31:
> * Having a big testcase that requires a separate download reduces the amount of people that will bother to look at your issue

This is true, but we also must keep in mind the overall dynamics of the issue. Fuzzer authors derive substantial benefit from fuzzer-found issues that are acknowledged by the project being fuzzed, such as publishing of academic papers and procurement of funding. Because of this they are incentivized to submit as many issues as they can in the lowest possible time, often not investigating the causes or minimizing the reproducer at all (e.g. this issue). Many such bugs are actually already known ones (fuzzers tend to have a very high rediscovery rate), and as a result, the project suffers from having to dedicate triage resources while often deriving no benefit from the report at all.

I see this pattern as exploitative and believe we must not encourage it.

georgerennie, 2024-12-31:
> This is true, but we also must keep in mind the overall dynamics of the issue

Maybe we are talking past each other here, I am totally in agreement and my points were meant as guidance towards what practical steps in submitting a fuzzer found bug make it easy to be quickly triaged/fixed if appropriate

whitequark, 2024-12-31:
I meant to add to what you said, not to contradict it, apologies.

FSY369, 2024-12-31:
Ok, thank you very much for your patient reply. This will not bother me, but will let me learn how to submit bugs, which will improve efficiency and achieve a win-win situation. Thank you again for your help and support.

---- Replied Message ----
| From | ***@***.***> |
| Date | 12/31/2024 18:36 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [YosysHQ/yosys] Inconsistent simulation before and after synthesis of complex assignment expressions (Issue #4828) |

I meant to add to what you said, not to contradict it, apologies.

FSY369, 2024-12-31:
thank you very much for your patient reply. This will not bother me, but will let me learn how to submit bugs, which will improve efficiency and achieve a win-win situation. Thank you again for your help and support.

---- Replied Message ----
| From | ***@***.***> |
| Date | 12/31/2024 18:36 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [YosysHQ/yosys] Inconsistent simulation before and after synthesis of complex assignment expressions (Issue #4828) |

I meant to add to what you said, not to contradict it, apologies.

FSY369, 2025-01-03:
[rel3.txt](https://github.com/user-attachments/files/18296852/rel3.txt)

whitequark, 2025-01-14:
This is still not minimized enough as your testbench includes a lot of random data and your `rel3.txt` file includes far too much code, including obviously extraneous constructs like `$unsigned($unsigned(...))`. 

You should:
1. Use tools like C-Reduce to further reduce the amount of code in the testcase.
2. Remove lines from your testbench until only a single line remains that triggers the issue.
3. Subset the `synth` script to determine which pass(es) are responsible for causing the mismatch.
4. Manually remove parts of the RTL until the module consists of only a few Verilog operators (the issue almost certainly can be reproduced with 1 or 2 operators).
5. Simplify the very large constants used in the testbench so that the values of inputs can be easily understood by a human without having to decipher code generated by your fuzzing tools.

FSY369, 2025-02-20:
I'm sorry, does that meet the requirements？

[rtl4.txt](https://github.com/user-attachments/files/18880339/rtl4.txt)

georgerennie, 2025-02-20:
This is reduced which is good, however it doesn't seem to show any issue. Rather than just reducing to an example that you expect to fail you should also try to work out what construct you think yosys is handling incorrectly. This means it is more likely that you submit a bug report for something that is a real bug rather than a bug in your simulation setup.

Simulating it with your testbench before synthesis returns a load of zeros followed by
```
zzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzxxxxxxxxxxxxxx0
```
for the first line, then a load of zeros followed by
```
zzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzz111111111111110
```
for following lines.

By comparison, simulating it after synthesis returns a load of zeros followed by
```
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx111111111111110
```
for all inputs.

Yosys is generally allowed to convert `x` to `1` and (typically) `z` to `x` so this seems to be a valid refinement.

FSY369, 2025-02-28:
[rtl5.txt](https://github.com/user-attachments/files/19019137/rtl5.txt)

georgerennie, 2025-02-28:
This doesn't even compile with iverilog
```
rtl5.v:13: error: Unable to bind wire/reg/memory `wire230' in `testbench.top1'
rtl5.v:17: error: Unable to bind wire/reg/memory `wire230' in `testbench.top1'
```

closed 2025-03-21 by KrystalDelusion