YosysHQ/yosys #4830: Support `gzip`ed Liberty files

hpretl, 2025-01-01:
### Feature Description

Happy New Year, YosysHQ team! Please keep up the excellent work you are doing!

It looks like (I haven't tested myself) that Yosys is already supporting gzip'ed Verilog files. It would be equally useful to support gzip'ed Liberty (`.lib.gz`) files, as having zipped Liberty files in the PDK setup would be a big help for space reduction.

See also https://github.com/efabless/openlane2/issues/627

donn, 2025-01-01:
Yosys does support gzipped liberty files, it's our Yosys script that emitted that error :)

hpretl, 2025-01-01:
Sounds good, then looks like we could have .gz support soon in OL2 :-)

povik, 2025-01-01:
Thanks and happy new year to you too!

> Yosys does support gzipped liberty files

It does support them for `read_liberty`, not for `abc -liberty` or `dfflibmap -liberty`. @widlarizer happens to be looking into this.

hpretl, 2025-01-01:
@widlarizer +1 for nickname :-)

hpretl, 2025-01-05:
OL2 is almost released with`.gz` support (planned for `2.3.2` version). @widlarizer Do you have a timeline for full `.gz` support rolled into a Yosys release?

donn, 2025-01-05:
We wouldn't include something as major as a Yosys upgrade in a patch version btw

gadfort, 2025-01-06:
`dfflibmap` and the `clockgate` command would also need to be able to support a dont use list, otherwise to enable that we still have to preprocess the file.

povik, 2025-01-06:
I think both do already (at least as of master)

gadfort, 2025-01-06:
@povik thanks for clarifying, I hadn't seen that

gadfort, 2025-01-06:
@povik is there a plan to support multiple liberty files on these commands as well? (like abc)?

povik, 2025-01-06:
Yes, there is. Btw as @widlarizer points out in the linked PR, for `abc -liberty` to accept gzipped Liberty files we need to change ABC, not Yosys. We don't have concrete plans to do that yet. If anyone in the community wants to pick this task up please let the Yosys team know.

gadfort, 2025-01-06:
@povik one possibility would be to extract the files into the tmp directory that abc is running in if the liberty file has a .gz extension (avoiding the need to update ABCs support)

povik, 2025-01-06:
Right, we already prepare a bunch of temp files for ABC. Though it's suboptimal for those PDKs where the Liberty files are excessively large but at least it gets us something in the short term.

hpretl, 2025-01-06:
@povik I think the approach with extract to `/tmp` would be a good compromise to get this in reasonable short time, and have mid-term `abc` prepared for `.gz` support.

widlarizer, 2025-01-07:
@gadfort 

> is there a plan to support multiple liberty files on these commands as well? (like abc)?

We're streets ahead of ya: #4797, sorry it didn't make it into the release notes!

gadfort, 2025-01-07:
@widlarizer even better :-)

povik, 2025-02-28:
abc gzip support has been added by @wjrforcyber in https://github.com/berkeley-abc/abc/pull/362

hpretl, 2025-03-12:
@povik @mmicko Can we close this since this has been merged? https://github.com/YosysHQ/yosys/commit/a7267f4f7bb6bf0aff235cb4cb8a80f918c68741

hpretl, 2025-03-12:
@mmicko Alright, I was under the impression that all the related PR are in :-) I'll hope the missing piece will be merged soon ;-)

PR #4834 (2025-03-20): Memory-efficient zlib usage across Liberty file consumers
In ASIC flows, Liberty files provided in the PDK can be very large and are often compressed with gzip. Prior to this PR, the only pass that accepted `.lib.gz` was `read_liberty`, just like `read_verilog` accepts `.v.gz` etc. `-liberty filename` arguments to passes like `clockgate` and `dfflibmap` didn't use the decompression code.

The `decompress_gzip` function also has been reading the entirety of the uncompressed file into RAM (into a stringstream). This is now fixed by implementing an `gzip_istream` much like the pre-existing `gzip_ [... cut]
closed 2025-03-20 by KrystalDelusion