YosysHQ/yosys #4844: synth failure

FSY369, 2025-01-14:
### Version

Yosys 0.48+47 (git sha1 cbb95cb51, clang++ 18.1.8 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

1. Enter yosys in the terminal
2. read the rtl.v file, read_verilog rtl.v
3. synthesize the command, synth
4. synthesis failed

[rtl.v.log](https://github.com/user-attachments/files/18405090/rtl.v.log)

### Expected Behavior

Synthesize Success

### Actual Behavior

synthesis failed，2.5. Executing CHECK pass (checking for obvious problems).
Checking module module128...
Checking module module150...
浮点数例外 (核心已转储)

whitequark, 2025-01-14:
You are submitting a bug report produced using fuzzing tools. With such bug reports, it is often hard to tell whether the issue reported is an issue in Yosys or an issue with the test harness.

You [were instructed](https://github.com/YosysHQ/yosys/issues/4828) how to submit such bug reports properly and you are not following the instructions. As such I am closing your report as invalid.

FSY369, 2025-01-14:
Sorry, I uploaded the wrong file, here is the minimal test case, the problem occurs with the assignment of reg161 thus causing the synthesis to fail.

[rtl.v.log](https://github.com/user-attachments/files/18405287/rtl.v.log)

whitequark, 2025-01-14:
Thanks. For other readers, it seems that:

> 浮点数例外 (核心已转储)

means

> Floating point exception (core dumped)

whitequark, 2025-01-14:
I have reduced your testcase further using [C-Reduce](https://github.com/csmith-project/creduce). Here is the actually minimal testcase:

```verilog
module a(b, c, d);
	output b;
	input c;
	input d;
	always b <= $signed(f) << (0 ? 0 : g);
endmodule
```

I have done this by putting your code into `issue4844.v`, writing a shell script `issue4844.sh` that checks for a floating point error (code 136) being returned:

```sh
#!/bin/sh
yosys issue4844.v -p synth
[ $? -eq 136 ]
```

and running C-Reduce as:

```console
$ creduce --not-c issue4844.sh issue4844.v
```

whitequark, 2025-01-14:
What happens is that in the following code:

https://github.com/YosysHQ/yosys/blob/6225abec71fe2cdc6e1fd32b727d9d4b9e3319fc/kernel/celledges.cc#L250-L257

`skip` becomes 0 as `k` becomes 31. This actually seems like it would happen pretty much every time there is a signed left shift by a 32-bit value; I'm surprised it doesn't trigger more often.

I'm not actually sure what exactly this code is doing so I'll let someone else fix it.

whitequark, 2025-01-14:
Oh also `1 << b_width` triggers UB on each iteration of the loop when `b_width == 32` (C99 §6.5.7.3):

> The integer promotions are performed on each of the operands. The type of the result is that of the promoted left operand. If the value of the right operand is negative or is greater than or equal to the width of the promoted left operand, the behavior is undefined.

whitequark, 2025-01-14:
@povik @nakengelhardt It looks like this bug was introduced in https://github.com/YosysHQ/yosys/pull/3972. Not to put too fine a point on it, but the nondeterminism that I complained about two years ago was caused by invoking UB, apparently, every single time that we had code akin to `$signed(1) << 1`, and it should have been investigated then instead of making excuses for it.

whitequark, 2025-01-14:
By setting `SANITIZER = undefined` in `Makefile.conf` and setting the variable `UBSAN_OPTIONS=halt_on_error=1` I am able to get it to crash in the sanitizer:

```
Checking module a...
kernel/celledges.cc:256:50: runtime error: shift exponent 32 is too large for 32-bit type 'int'
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior kernel/celledges.cc:256:50 in 
```

I think we should run ubsan on CI and I've filed https://github.com/YosysHQ/yosys/pull/4845 to do that.

I ran the testsuite with UBSan enabled and got this:

```
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior frontends/ast/simplify.cc:2939:21 in
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior kernel/celledges.cc:254:19 in
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior kernel/celledges.cc:255:22 in
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior kernel/celledges.cc:256:50 in
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior kernel/celledges.cc:259:30 in
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior kernel/celledges.cc:262:30 in
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior libs/fst/fastlz.cc:227:5 in
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior libs/fst/fastlz.cc:227:5 in
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior libs/fst/fastlz.cc:370:5 in
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior libs/fst/fastlz.cc:370:5 in
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior libs/fst/fastlz.cc:372:5 in
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior libs/fst/fastlz.cc:372:5 in
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior libs/fst/fastlz.cc:512:11 in
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior libs/fst/fastlz.cc:512:18 in
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior libs/fst/fastlz.cc:513:11 in
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior libs/fst/fastlz.cc:513:18 in
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior libs/fst/fastlz.cc:514:11 in
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior libs/fst/fastlz.cc:514:18 in
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior libs/fst/fastlz.cc:515:11 in
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior libs/fst/fastlz.cc:515:18 in
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior libs/fst/fastlz.cc:518:11 in
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior libs/fst/fastlz.cc:518:18 in
```

Aside from the issue in astsimplify introduced in 2cab4ff173, all of the instances of UB are either in `celledges.cc`, or in `fastlz.cc` for unaligned loads/stores that we should probably add to the exclusion list.

FSY369, 2025-01-15:
So, it's really a flaw in yosys isn't it?

whitequark, 2025-01-15:
Yes, definitely, and a fairly important one in my opinion.

georgerennie, 2025-03-20:
The patches didn't fix all the UB here as shown in #4947. The issue with skip becoming 0 as K > 31 still persists, and it causes division by 0 in `i % skip`. UBSAN still detects this for the testcase in this thread, or the minimized testcase below from #4947. These should be added as testcases when fixing this so we don't just see them again.

```
read_rtlil << EOF
attribute \top 1
module \top
  wire output 1 \a
  cell $shl \cell
    parameter \A_SIGNED 1
    parameter \A_WIDTH 1
    parameter \B_SIGNED 0
    parameter \B_WIDTH 49
    parameter \Y_WIDTH 1
    connect \A 1'x
    connect \B 49'x
    connect \Y \a
  end
end
EOF
check
```

```
kernel/celledges.cc:255:22: runtime error: signed integer overflow: -2147483648 - 1 cannot be represented in type 'int'
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior kernel/celledges.cc:255:22
kernel/celledges.cc:254:19: runtime error: shift exponent 32 is too large for 32-bit type 'int'
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior kernel/celledges.cc:254:19
```

KrystalDelusion, 2025-03-20:
Ah, so #4879 only fixed for the case where `k == 31`, dang.

georgerennie, 2025-03-20:
I'm not sure it fixed it for that case even, as `1 << 31 + 1` is going to overflow an int

KrystalDelusion, 2025-03-20:
Ah, I was testing with `tests/simple/memory.v` because that was the one already in CI and failing with
```
kernel/celledges.cc:262:30: runtime error: shift exponent 32 is too large for 32-bit type 'int'
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior kernel/celledges.cc:262:30 
```
which is one of the cases that doesn't use k, so I fixed `b_width >= 31`, but not `k >= 31`.

KrystalDelusion, 2025-05-11:
Looks like this was fixed in #5066.

PR #4879 (2025-02-03): Fixing undefined behaviours
_What are the reasons/motivation for this change?_

#4845 (and https://github.com/YosysHQ/yosys/issues/4844#issuecomment-2589140999) reveal undefined behaviors in the Yosys code.  This PR aims to resolve them.

_Explain how this is achieved._

- [x] kernel/celledges.cc
`b_width_capped` already exists for preventing arithmetic overflow, limiting the value of `b_width` to 30.  This just changes the left shifts to also use it. The caveat of incorrect results for extremely large values of `a_width` still applies, as does the improbability of that actually happening. Thi [... cut]
commit 31b00b4c (2025-02-03): celledges: Use b_width_capped for left shifts

`b_width_capped` already exists for preventing arithmetic overflow, limiting the value of `b_width` to 30.  This just changes the left shifts to also use it.
The caveat of incorrect results for extremely large values of `a_width` still applies, as does the improbability of that actually happening.
This fixes #4844 (or at least, the floating point exception; the circuit still isn't valid but I think that's fine).
closed 2025-05-11 by KrystalDelusion