YosysHQ/yosys #4853: Yosys mistakenly detects a combinational loop in the btor backend

flaviens, 2025-01-16:
### Version

Yosys 0.48+77 (git sha1 eac2294ca, g++ 12.2.0-19 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

Hi there!

Yosys mistakenly identifies a combinational loop when executing the btor backend on a miter of the CVA6 FPU controller. Please find the steps to reproduce this issue.

The Yosys script that I use is
```
yosys read_verilog -sv control_mvp.sv
yosys miter -equiv control_mvp control_mvp miter_control_mvp
yosys hierarchy -top miter_control_mvp
yosys prep -top miter_control_mvp
yosys flatten
yosys opt
yosys async2sync
yosys dffunmap
yosys write_btor cva6.btor
```

[Source](https://github.com/user-attachments/files/18444596/control_mvp.zip)

If I do not create a miter circuit, then Yosys does not report a combinational loop, and by construction a miter cannot create one that does not exist in the original circuit.

Thanks!

### Expected Behavior

The backend should execute correctly.

### Actual Behavior

Yosys fails during the btor backend: [Resulting log](https://github.com/user-attachments/files/18444587/control_mvp.log)

georgerennie, 2025-01-16:
This isn't due to the miter, its a loop Yosys is detecting in the original sources. Consider the following script that fails:
```
read_verilog -sv control_mvp.sv
prep -top control_mvp -flatten
scc -expect 0
```

Running `synth` the path disappears, so its probably a spurious path that only exists at the word level. The btor backend requires there to be no word level loops so I'd suggest there probably isn't a Yosys bug here and the easiest solution is likely to track down the path and see if you can modify the sources slightly so the signals are broken up in such a way that the path doesnt exist.

whitequark, 2025-01-16:
There is also `splitnets -driver`.

georgerennie, 2025-01-16:
> There is also splitnets -driver.

I had a quick try with one or two tricks like this and none seemed to solve this particular instance. One issue with splitnets for this is that it won't help if breaking the path needs breaking up a cell (which I believe it does in this case)

At least one path seems to be

`Sqrt_Q1` -> `Iteration_cell_b_BMASK_D[1]` [src](https://github.com/pulp-platform/fpu_div_sqrt_mvp/blob/86e1f558b3c95e91577c41b2fc452c86b04e85ac/hdl/control_mvp.sv#L2284)
`Iteration_cell_b_BMASK_D[1]` -> `Iteration_cell_b_D[1]` [src](https://github.com/pulp-platform/fpu_div_sqrt_mvp/blob/86e1f558b3c95e91577c41b2fc452c86b04e85ac/hdl/control_mvp.sv#L2346C22-L2346C40)
`Iteration_cell_b_D[1]` -> `Iteration_cell_carry_D[1]` [src1](https://github.com/pulp-platform/fpu_div_sqrt_mvp/blob/86e1f558b3c95e91577c41b2fc452c86b04e85ac/hdl/control_mvp.sv#L2353) [src2](https://github.com/pulp-platform/fpu_div_sqrt_mvp/blob/86e1f558b3c95e91577c41b2fc452c86b04e85ac/hdl/iteration_div_sqrt_mvp.sv#L59)
`Iteration_cell_carry_D[1]` -> `Sqrt_quotinent_S` [src](https://github.com/pulp-platform/fpu_div_sqrt_mvp/blob/86e1f558b3c95e91577c41b2fc452c86b04e85ac/hdl/control_mvp.sv#L861)
`Sqrt_quotinent_S` -> `Sqrt_Q1` [src](https://github.com/pulp-platform/fpu_div_sqrt_mvp/blob/86e1f558b3c95e91577c41b2fc452c86b04e85ac/hdl/control_mvp.sv#L1407)

It seems this isn't a legitimate path. One way of breaking it seems to just be to do the assignment for each bit separately. In fact, the way Yosys collects these together you literally just need there to be somewhere in the always block where the bits are set individually and Yosys will do the mux inference for them each with a different cell.

It's a silly solution but the following diff in the verilog stops the SCCs so your original script passes:
```diff
diff --git a/old.sv b/control_mvp.sv
index 482326f..5b207e7 100644
--- a/old.sv
+++ b/control_mvp.sv
@@ -507,6 +507,10 @@ module control_mvp (
                case (Format_sel_S)
                        2'b00: begin
                                Sqrt_quotinent_S = {~Iteration_cell_sum_AMASK_D[0][28], ~Iteration_cell_sum_AMASK_D[1][28], ~Iteration_cell_sum_AMASK_D[2][28], ~Iteration_cell_sum_AMASK_D[3][28]};
+                               Sqrt_quotinent_S[3] = Sqrt_quotinent_S[3];
+                               Sqrt_quotinent_S[2] = Sqrt_quotinent_S[2];
+                               Sqrt_quotinent_S[1] = Sqrt_quotinent_S[1];
+                               Sqrt_quotinent_S[0] = Sqrt_quotinent_S[0];
                                Q_sqrt_com_0 = {{29 {1'b0}}, ~Q_sqrt0[28:0]};
                                Q_sqrt_com_1 = {{29 {1'b0}}, ~Q_sqrt1[28:0]};
                                Q_sqrt_com_2 = {{29 {1'b0}}, ~Q_sqrt2[28:0]};
```

I'm going to close this issue for now as I don't think this is really a Yosys issue, its a known limitation when dealing with comb loops at the word level.

closed 2025-01-16 by georgerennie

flaviens, 2025-01-16:
Thanks a lot!