YosysHQ/yosys #4865: Yosys synth_ice40 -dsp fails on an assertion

zapta, 2025-01-23:
### Version

Yosys 0.49+1 (git sha1 3076803c9, aarch64-apple-darwin23.5-clang++ 18.1.8 -fPIC -O3)

### On which OS did this happen?

macOS

### Reproduction Steps

* Unzip the zip file below
* Source oss-cad-suite ``environment``
* Run build.sh

[yosys-ice40-dsp-error.zip](https://github.com/user-attachments/files/18524355/yosys-ice40-dsp-error.zip)

main.v
```
module main (

    input  clki,
    input  user_1,  // serial data in.
    output reg    user_2,  // serial data out

    // USB Pins (which should be statically driven if not being used).
    output usb_dp,
    output usb_dn,
    output usb_dp_pu
);

  // Prevent the host from trying to connect to this design.
  assign usb_dp = 1'b0;
  assign usb_dn = 1'b0;
  assign usb_dp_pu = 1'b0;

  // The size of each of the serial registers.
  // TODO: Why this fails with N = 40.
  //
  // localparam N = 38;
  localparam N = 40;
  // localparam N = 42;

  // The two words we multiply.
  reg [N-1:0] reg1;
  reg [N-1:0] reg2;

  always @(posedge clki) begin
    // Shift din into 2 x N words 
    reg1   <= {reg1[N-2:0], user_1};
    reg2   <= {reg1[N-2:0], reg1[N-1]};
    // Multiply the words, and output the parity of the result.
    user_2 <= ~^(reg1 * reg2);
  end

endmodule
```

Entire example:

### Expected Behavior

Yosys completes successfully and the designs uses the ICESTORM_DSP cells.

### Actual Behavior

Yosys fails on an assertion

```
⦗OSS CAD Suite⦘ ~/work$ ./build.sh 
+ yosys --version
Yosys 0.49+1 (git sha1 3076803c9, aarch64-apple-darwin23.5-clang++ 18.1.8 -fPIC -O3)
+ yosys -p 'synth_ice40 -top main -dsp -json _build/hardware.json' -q main.v
ERROR: Assert `nusers(O.extract_end(i)) <= 1' failed in ./passes/pmgen/ice40_dsp_pm.h:388.
⦗OSS CAD Suite⦘ ~/work$ 
```

widlarizer, 2025-01-28:
Reduced
```v
module main (
    input  clki,
    output reg user_2
);

  // fails only for some widths of these with no obvious pattern
  localparam N = 34;
  reg [N-1:0] reg1;
  reg [N-1:0] reg2;

  always @(posedge clki) begin
    user_2 <= ^(reg1 * reg2);
  end

endmodule
```

widlarizer, 2025-01-28:
Isolated RTLIL before the `ice40_dsp` pass: [usb-bugpoint.txt](https://github.com/user-attachments/files/18577911/usb-bugpoint.txt). `yosys -p 'read_rtlil usb-bugpoint.txt; ice40_dsp'`. The pass fails on a SB_MAC16 that has a range of multiple-user bits starting at bit 0, as well as some other bit higher up. This higher bit triggers the assertion failure.
```
bit 0 users 2
bit 1 users 2
bit 2 users 1
bit 3 users 1
bit 4 users 1
...
bit 15 users 1
bit 16 users 2 <- problem!
bit 17 users 2 <- problem!
bit 18 users 1
bit 19 users 1
...
bit 31 users 1
```

closed 2025-04-23 by KrystalDelusion

zapta, 2025-04-23:
Thanks @Anhijkt !!!