YosysHQ/yosys #4869: level of non-const reset logged incorrectly? after #4569

donn, 2025-01-28:
### Version

0.49

### On which OS did this happen?

macOS

### Reproduction Steps

mkdir and cd into an empty directory, where you download and untar this archive:

[yosys_repro.tgz](https://github.com/user-attachments/files/18571165/yosys_repro.tgz)

Then `/path/to/yosys run.ys`.

### Expected Behavior

`proc_dff` (run as part of `synth`) reports the following:

```
3.2.9. Executing PROC_DFF pass (convert process syncs to FFs).
Creating register for signal `\non_const_reset.\value' using process `\non_const_reset.$proc$non_const_reset.v:9$2'.
Warning: Async reset value `\rstval' is not constant!
  created $aldff cell `$procdff$8' with positive edge clock and positive level non-const reset.
```

However, the asynchronous reset in question is negative:

```v
module non_const_reset(
    input clk,
    input rstn,
    input rstval,
    output reg value
);
    wire value_next = ~value;

    always_ff @ (posedge clk or negedge rstn)
        if (!rstn)
            value <= rstval;
        else
            value <= value_next;
endmodule
```

Yet, with the provided mapping file, the result from synthesis is identical to the expected behavior. I've provided the logs from both a known-working version and 0.49 in the reproducible tarball.

To be 100% clear, I'm not sure if I'm misunderstanding something or if this is an actual logging bug.

### Actual Behavior

The issue bisects to #4569 (i.e. Yosys 0.44+60), where before that patch the report is as follows:

```
3.2.9. Executing PROC_DFF pass (convert process syncs to FFs).
Creating register for signal `\non_const_reset.\value' using process `\non_const_reset.$proc$non_const_reset.v:9$2'.
Warning: Async reset value `\rstval' is not constant!
  created $aldff cell `$procdff$4' with positive edge clock and negative level non-const reset.
```

It's worth mentioning that there was a separate issue with this patch when there are multiple asynchronous resets that was fixed in #4714.

povik, 2025-01-28:
FYI @georgerennie @widlarizer

widlarizer, 2025-01-28:
In both cases, the `$aldff` emitted by `proc_dff` itself has the same polarity. In the negedge case, there is an extra `$not` cell emitted. The log then correctly represents what happens inside Yosys, which as you show, leads to correct synthesis in the context of the whole flow. There might be more to this issue but I don't have the capacity to dig deeper at the moment

widlarizer, 2025-01-28:
[negedge.log](https://github.com/user-attachments/files/18572171/negedge.log)
[posedge.log](https://github.com/user-attachments/files/18572172/posedge.log)

donn, 2025-01-28:
Thanks for explaining!

closed 2025-01-28 by donn