YosysHQ/yosys #4872: Yosys will sometimes emit a flip-flop that never toggles instead of a constant 0

leocassarani, 2025-01-28:
### Version

Yosys 0.49+3 (git sha1 3d35f367c, x86_64-apple-darwin23.5-clang++ 18.1.8 -fPIC -O3)

### On which OS did this happen?

macOS

### Reproduction Steps

I was seeing some very strange behaviour as a result of Yosys being able to optimise away large parts of my design. Eventually I was able to produce a minimal reproduction case, which uncovered a [nextpnr bug](https://github.com/YosysHQ/nextpnr/pull/1442).

When running `synth_ecp5` on this Verilog file, Yosys will output a useless flip-flop with `CE` set to 1 and `CEMUX` set to `INV`. Similarly, `synth_ice40` produces a flip-flop with `E` set to 0 and `D` set to 1.

This is because Yosys correctly recognises that `b` will never be 0, and therefore `c` will never be set to 1.

```verilog
module top (
  input clk,
  output reg c = 1'b0
);

  reg a = 1'b0;
  always @(posedge clk)
    a <= !a;

  wire [2:0] b = 3'd3 + { 2'b0, a };

  always @(posedge clk)
    if (b == 3'd0) c <= 1'b1;

endmodule
```

However, if `b` is changed to any other 3-bit value apart between 1 and 6 inclusive, Yosys correctly optimises away the flip-flop and outputs a constant 0.

Calling `synth_ecp5/ice40` with the `-dff` flag produces the desired result, but it's not clear to me why that should be necessary in this simple case.

Note that this behaviour is also present for bit widths larger than 3. For example, this variation also produces a flip-flop that will never trigger (while values lower than `5'd13` correctly output a constant 0):

```verilog
module top (
  input clk,
  output reg c = 1'b0
);

  reg [1:0] a = 2'd0;
  always @(posedge clk)
    a <= a + 1;

  wire [4:0] b = 5'd13 + { 3'b0, a };

  always @(posedge clk)
    if (b == 5'd0) c <= 1'b1;

endmodule
```

### Expected Behavior

```
module \top

  wire output 2 \c

  wire input 1 \clk

  connect \c 1'0
end
```

### Actual Behavior

## `synth_ecp5`

```
module \top

  attribute \unused_bits "0 1 2"
  wire width 3 \b

  wire output 2 \c

  wire input 1 \clk

  attribute \module_not_derived 1
  cell \TRELLIS_FF \c_TRELLIS_FF_Q
    parameter \CEMUX "INV"
    parameter \CLKMUX "CLK"
    parameter \GSR "DISABLED"
    parameter \LSRMUX "LSR"
    parameter \REGSET "RESET"
    connect \CE 1'1
    connect \CLK \clk
    connect \DI 1'1
    connect \LSR 1'0
    connect \Q \c
  end

  connect \b [0] \b [1]
end
```

## `synth_ice40`

```
module \top

  wire width 3 \b

  wire output 2 \c

  wire input 1 \clk

  attribute \module_not_derived 1
  cell \SB_DFFE \c_SB_DFFE_Q
    connect \C \clk
    connect \D 1'1
    connect \E 1'0
    connect \Q \c
  end

  connect \b 3'x
end
```

widlarizer, 2025-02-04:
Thanks for the report!

This is a missed optimization when you don't use `-dff` for `synth_*`. I don't consider that a bug, it's a separate thing. You're presenting it as inconsistent with cases where `b` has a constant value, which is a way easier thing to detect - at that point, the `$alu` cell is fully driven by constants and evaluates to a constant and so does the `CE` port for the `$dffe`, causing the `$dffe` to go away. Propagating possible ranges of values through the design to detect that a bit boiled down from arithmetics is always reset is probably infeasible and wouldn't work right without using an SMT solver due to diamond patterns. Propagating constants is way simpler. So the optimizations in yosys are doing their best within their scope and I don't believe this is really inconsistent.

Later, `abc` has more information when actually synthesizing to LUT and catches if it's allowed to see flops when `-dff` is passed. @Ravenslofty do you happen to know why `-dff` is not on by default for `abc9` etc on the side of yosys? Run time concerns?

leocassarani, 2025-02-04:
Hey @widlarizer, thanks for looking into this.

> You're presenting it as inconsistent with cases where `b` has a constant value, which is a way easier thing to detect - at that point, the `$alu` cell is fully driven by constants and evaluates to a constant and so does the `CE` port for the `$dffe`, causing the `$dffe` to go away.

My apologies for the confusion, I've re-read my report and I should have been clearer about the inconsistency. To remove any misunderstanding, I'm saying that this results in a never-triggering flip-flop:

```verilog
wire [2:0] b = 3'd3 + { 2'b0, a };
```

But all of the following (correctly) optimise the flip-flop away:

```verilog
wire [2:0] b = 3'd1 + { 2'b0, a };
wire [2:0] b = 3'd2 + { 2'b0, a };
wire [2:0] b = 3'd4 + { 2'b0, a };
wire [2:0] b = 3'd5 + { 2'b0, a };
wire [2:0] b = 3'd6 + { 2'b0, a };
```

And as I show later in my report, there are similarly pathological values for wider bit widths (e.g. `5'd13`). I thought this behaviour was very odd, so it seemed worth reporting as there's clearly something unexpected going on for specific bit patterns.