YosysHQ/yosys #4873: 'show -format dot' fails on a .sv file

zapta, 2025-01-29:
### Version

Yosys 0.49+3 (git sha1 9d4630484, x86_64-apple-darwin23.5-clang++ 18.1.8 -fPIC -O3)

### On which OS did this happen?

macOS

### Reproduction Steps

1. Unzip the zip file below in an empty directory.

2. Run the command below. It succeeds.
```
⦗OSS CAD Suite⦘ ~/work$ yosys -p "synth_ice40 -top and_gate -json hardware.json" -q and_gate.sv
⦗OSS CAD Suite⦘ ~/work$ 
```

3. Run the command below. It fails.
```
⦗OSS CAD Suite⦘ ~/work$ yosys -f verilog -p "show -format dot -colors 1 -prefix hardware and_gate" -q and_gate.sv
and_gate.sv:2: ERROR: syntax error, unexpected TOK_ID, expecting ',' or '=' or ')'
⦗OSS CAD Suite⦘ ~/work$ 
```

[work.zip](https://github.com/user-attachments/files/18582074/work.zip)

Source file:
```
module and_gate (
    input  logic a,
    input  logic b,
    output logic s
);

  assign s = a & b;

endmodule
```

### Expected Behavior

The command should succeed and generate a .dot file with the graph.

### Actual Behavior

The command fails.

```
⦗OSS CAD Suite⦘ ~/work$ yosys -f verilog -p "show -format dot -colors 1 -prefix hardware and_gate" -q and_gate.sv
and_gate.sv:2: ERROR: syntax error, unexpected TOK_ID, expecting ',' or '=' or ')'
⦗OSS CAD Suite⦘ ~/work$ 
```

closed 2025-01-29 by georgerennie

zapta, 2025-01-30:
Thanks George. It works.

povik, 2025-02-06:
> I will also note that as for the discussion of why yosys-slang isn't working in the mentioned apio issue, there is a slang option to ignore unknown modules that may help:
> `yosys -m slang -p "read_slang --ignore-unknown-modules leds.v main.v oscillator.v; show -format dot -colors 1 -prefix _build/hardware main" -q`

This can introduce subtle correctness issues in elaboration of the design since the frontend doesn't know the direction and sizes of ports on blackboxes and instead goes by best guesses. That's why I don't recommend using the option to anyone and I may need to remove it in the future.

> This is the default behaviour for Yosys' built in verilog frontend.

It is, but the fact that the elaboration of modules isn't informed by definitions (even when they are available) leads the bugs. If you elaborate

```
module m(input [7:0] g);
endmodule

module top;
	m f(.g(~2'd3));
endmodule
```

with the built-in frontend, the `g` port will be connected to `8'd0`, which isn't the correct answer. yosys-slang is architected to do better.

zapta, 2025-02-06:
Hi @povik, below are the commands we used in Apio to build and to graph projects, regardless if it has ``.v`` or ``.sv`` files. Are they reasonable?

Build
```
yosys -p "synth_gowin -top blinky -json _build/hardware.json" -q blinky.sv
nextpnr-himbaechel --device GW1NR-LV9QN88PC6/I5 --json _build/hardware.json --write _build/hardware.pnr.json --report _build/hardware.pnr --vopt family=GW1N-9C --vopt cst=blinky.cst -q
gowin_pack -d GW1N-9C -o _build/hardware.fs _build/hardware.pnr.json
```

Graph
```
yosys -p "read_verilog -sv blinky.sv; show -format dot -colors 1 -prefix _build/hardware blinky" -q
dot -Tsvg _build/hardware.dot -o _build/hardware.svg
```

povik, 2025-02-06:
Hi @zapta, those seem to be reasonable commands if you are using the built-in frontend, i.e. read_verilog. As an alternative there's read_slang with better SV support, but it's also harder to set up for FPGA targets. Some discussion of what it would take was held at povik/yosys-slang#71

povik, 2025-02-06:
Wrong link, I've meant to link povik/yosys-slang#72

zapta, 2025-02-07:
Thanks @povik, please ping us here or on the other thread when there will be a better choice of yosys commands for apio.