YosysHQ/yosys #4882: `read_verilog` never completes and continues to grow system memory usage

gadfort, 2025-02-01:
### Version

Yosys 0.49+10 (git sha1 18a7c0038, g++ 11.4.0-1ubuntu1~22.04 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

[test.tar.gz](https://github.com/user-attachments/files/18626734/test.tar.gz)

```
tar xvf test.tar.gz
cd test
yosys -s run.ys
```

### Expected Behavior

Completes reading, ideally without a large memory footprint.

### Actual Behavior

```
-- Executing script file `run.ys' --

1. Executing Verilog-2005 frontend: fdivsqrtuslc4.v
Parsing SystemVerilog input from `fdivsqrtuslc4.v' to AST representation.
Generating RTLIL representation for module `\fdivsqrtuslc4'.
Warning: Replacing memory \USel4 with list of registers. See fdivsqrtuslc4.v:44
```

It never finishes replacing the memory for USel4 and continues to grow in system memory (I killed it at 120GB).

Looking at a stacktrace it appears to be stuck in: `mem2reg_as_needed_pass2` (maybe helpful, maybe not)

widlarizer, 2025-02-17:
Minimized:
```v
module fdivsqrtuslc4 (
);
	reg [3:0] mem [1023:0];
	always @(*) begin
		integer w;
		integer i;
		i = 1;
		for (w = 0; w < 8; w = w + 1)
			begin
				mem[i] = 1'b0;
			end
	end
endmodule
```
Runtime and memory consumption scales with the upper bound of `w`
```
w time[s]  mem[MB]
1    3.30  428.18
2    9.21  805.88
4   28.82 1561.82
8   98.65 3073.55
```
Tripling of the runtime and doubling of the memory for each doubling of the iteration count might be ok, but memory spent even for just one loop iteration is kind of extreme. It's consistent with poor performance of `read_verilog` overall, so it's unlikely to be easily fixed in the near future at some root cause

marzoul, 2025-03-31:
This looks like a similar issue than https://github.com/YosysHQ/yosys/issues/4882
There is no issue with the Slang plugin for Yosys, which bypasses the Yosys verilog parser and frontend.
Would be worth a try here.

infinitymdm, 2025-03-31:
> This looks like a similar issue than [#4882](https://github.com/YosysHQ/yosys/issues/4882) There is no issue with the Slang plugin for Yosys, which bypasses the Yosys verilog parser and frontend. Would be worth a try here.

If you take a look at the linked scgallery issue, that's exactly what we ended up doing. It's still something that probably ought to get resolved on the Yosys frontend though.