YosysHQ/yosys #4883: Regression. Segmentation fault at commit 901935fbcec42cbf79594efd9cc2877877810395715

yrabbit, 2025-02-02:
### Version

Yosys 0.49+10 (git sha1 18a7c00382cd64d46b61ff6cafe80851ca29cb77, c++ 8.3 -O2 -fno-strict-aliasing -fPIC -O3)

### On which OS did this happen?

Linux, BSD

### Reproduction Steps

Compile the attached test design

``` shell
yosys -p "read_verilog ram.v; synth_gowin -json ram-synth.json"
```

test file:

[ram.v.txt](https://github.com/user-attachments/files/18630656/ram.v.txt)

### Expected Behavior

Error free compilation as it was up to "bad" commit https://github.com/YosysHQ/yosys/commit/901935fbcec42cbf79594efd9cc2877810395715

### Actual Behavior

Segmentation fault (core dumped)

```
2.8.14.9. Executing OPT_CLEAN pass (remove unused cells and wires).
Finding unused cells or wires in module \top..
Removed 4 unused cells and 4 unused wires.
<suppressed ~5 debug messages>

2.8.14.10. Executing MEMORY_COLLECT pass (generating $mem cells).

2.8.15. Executing OPT_CLEAN pass (remove unused cells and wires).
Finding unused cells or wires in module \top..

2.9. Executing MEMORY_LIBMAP pass (mapping memories to cells).
Segmentation fault (core dumped)
```

povik, 2025-02-02:
@widlarizer FYI

widlarizer, 2025-02-03:
Thanks for the quality report. The commit you bisected to seems to only expose the issue though since I can detect memory problems with valgrind on the reproducer with earlier versions as well
```
$ valgrind ./uut/0.45/bin/yosys -p "read_verilog garbage/ram.v; synth_gowin -json ram-synth.json" >/dev/null
==32735== Memcheck, a memory error detector
==32735== Copyright (C) 2002-2024, and GNU GPL'd, by Julian Seward et al.
==32735== Using Valgrind-3.24.0 and LibVEX; rerun with -h for copyright info
==32735== Command: ./uut/0.45/bin/yosys -p read_verilog\ garbage/ram.v;\ synth_gowin\ -json\ ram-synth.json
==32735==
==32735== Invalid read of size 4
==32735==    at 0xA075A5: bool Yosys::ModWalker::get_drivers<Yosys::hashlib::pool<Yosys::RTLIL::SigBit, Yosys::hashlib::hash_ops<Yosys::RTLIL::SigBit> > >(Yosys::hashlib::pool<Yosys::ModWalker::PortBit, Yosys::hashlib::hash_ops<Yosys::ModWalker::PortBit> >&, Yosys::hashlib::pool<Yosys::RTLIL::SigBit, Yosys::hashlib::hash_ops<Yosys::RTLIL::SigBit> > const&) const (in /home/emil/pulls/yosys/uut/0.45/bin/yosys)
==32735==    by 0xA015AF: Yosys::QuickConeSat::prepare() (in /home/emil/pulls/yosys/uut/0.45/bin/yosys)
==32735==    by 0xEE8492: (anonymous namespace)::MemMapping::assign_rd_ports() (in /home/emil/pulls/yosys/uut/0.45/bin/yosys)
==32735==    by 0xEDBFBD: (anonymous namespace)::MemMapping::MemMapping((anonymous namespace)::MapWorker&, Yosys::Mem&, Yosys::MemLibrary::Library const&, (anonymous namespace)::PassOptions const&) (in /home/emil/pulls/yosys/uut/0.45/bin/yosys)
==32735==    by 0xED8655: (anonymous namespace)::MemoryLibMapPass::execute(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, Yosys::RTLIL::Design*) (in /home/emil/pulls/yosys/uut/0.45/bin/yosys)
==32735==    by 0x8AB58B: Yosys::Pass::call(Yosys::RTLIL::Design*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >) (in /home/emil/pulls/yosys/uut/0.45/bin/yosys)
==32735==    by 0x8AAD3B: Yosys::Pass::call(Yosys::RTLIL::Design*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) (in /home/emil/pulls/yosys/uut/0.45/bin/yosys)
==32735==    by 0x8AC8D1: Yosys::ScriptPass::run(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) (in /home/emil/pulls/yosys/uut/0.45/bin/yosys)
==32735==    by 0x17CD49A: (anonymous namespace)::SynthGowinPass::script() (in /home/emil/pulls/yosys/uut/0.45/bin/yosys)
==32735==    by 0x17CC77B: (anonymous namespace)::SynthGowinPass::execute(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, Yosys::RTLIL::Design*) (in /home/emil/pulls/yosys/uut/0.45/bin/yosys)
==32735==    by 0x8AB58B: Yosys::Pass::call(Yosys::RTLIL::Design*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >) (in /home/emil/pulls/yosys/uut/0.45/bin/yosys)
==32735==    by 0x8AAD3B: Yosys::Pass::call(Yosys::RTLIL::Design*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) (in /home/emil/pulls/yosys/uut/0.45/bin/yosys)
==32735==  Address 0x52f7088 is 72 bytes inside a block of size 192 free'd
==32735==    at 0x4846786: operator delete(void*) (in /nix/store/c36qc6h4vzlyk5khfn0z7pfhjhxh0p9r-valgrind-3.24.0/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==32735==    by 0x8E64B4: Yosys::RTLIL::Module::remove(Yosys::RTLIL::Cell*) (in /home/emil/pulls/yosys/uut/0.45/bin/yosys)
==32735==    by 0xA09CDB: Yosys::Mem::remove() (in /home/emil/pulls/yosys/uut/0.45/bin/yosys)
==32735==    by 0xEE4C71: (anonymous namespace)::MemMapping::emit((anonymous namespace)::MemConfig const&) (in /home/emil/pulls/yosys/uut/0.45/bin/yosys)
==32735==    by 0xED861D: (anonymous namespace)::MemoryLibMapPass::execute(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, Yosys::RTLIL::Design*) (in /home/emil/pulls/yosys/uut/0.45/bin/yosys)
==32735==    by 0x8AB58B: Yosys::Pass::call(Yosys::RTLIL::Design*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >) (in /home/emil/pulls/yosys/uut/0.45/bin/yosys)
==32735==    by 0x8AAD3B: Yosys::Pass::call(Yosys::RTLIL::Design*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) (in /home/emil/pulls/yosys/uut/0.45/bin/yosys)
==32735==    by 0x8AC8D1: Yosys::ScriptPass::run(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) (in /home/emil/pulls/yosys/uut/0.45/bin/yosys)
==32735==    by 0x17CD49A: (anonymous namespace)::SynthGowinPass::script() (in /home/emil/pulls/yosys/uut/0.45/bin/yosys)
[... cut ...] mil/pulls/yosys/uut/0.45/bin/yosys)
==32735==    by 0x177F343: (anonymous namespace)::SynthPass::script() (in /home/emil/pulls/yosys/uut/0.45/bin/yosys)
==32735==    by 0x177E0C8: (anonymous namespace)::SynthPass::execute(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, Yosys::RTLIL::Design*) (in /home/emil/pulls/yosys/uut/0.45/bin/yosys)
(repeating)
```

Anhijkt, 2025-02-10:
The issue seems to be with Yosys::ModWalker::PortBit constructor. Since RTLIL::Cell cant be copied just by calling [constructor](https://github.com/YosysHQ/yosys/blob/main/kernel/rtlil.h#L1712) , [hash_into](https://github.com/YosysHQ/yosys/blob/main/kernel/modtools.h#L324) cant use the actual cell->name.

closed 2025-02-17 by yrabbit